Seeed-Studio/Wio_Link #2: how can I get the node token via the restful api

jiangzidong, 2016-01-16:
Hi,
How can I get the node token via the API in http://seeed-studio.github.io/Wio_Link/#status-code-and-format-of-the-response. I can't find one.

As for user token, i think `POST /v1/user/create`, `POST /v1/user/changepassword` and `POST /v1/user/login` will return the user token.

KillingJacky, 2016-01-17:
Hi Jiang,

Please move a little bit down on that page, the detailed response of each API call is illustrated on the right side of that API section.

KillingJacky, 2016-01-17:
The node token can be queried with /v1/nodes/list API http://seeed-studio.github.io/Wio_Link/#list-all-nodes-of-a-user. The key for node token is "node_key", sorry for the confusedness. And thanks for the reminder, we will add descriptions about the node token.

jiangzidong, 2016-01-17:
I see, thanks

closed 2016-01-17 by jiangzidong