Seeed-Studio/Wio_Link #4: Failing to configure

lawrie, 2016-03-30:
Hi, I am having difficulty configuring my Wio links. After many tries one of them is now configured, but I have tried 20 or 30 times and the other one will not configure. I always get the message saying Wio Link cannot connect to the router. I am providing the correct password. 

The Wio Link stays in "breathing mode". There is no other activity on the LEDs, so it looks like it is not getting the configuration message.

I have tried connecting a console to the Wio link UART and seeing if there are any diagnostics. I see the start-up diagnostic messages. In config mode the baud rate is 115200. I then see no other messages Is there any way of getting and diagnostic information to see what is going wrong?

My phone is an Android Moto G running Android 6.0.

KillingJacky, 2016-03-31:
Hi Lawrie

Sorry for the inconvenience. From the description, I agree with you that the Wio Link doesn’t even recv the config message. I’m sending you a firmware with debugging message enabled, please re-flash the firmware with the guide here: https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link <https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link> the only diff is to use user1.bin & user2.bin in my email attachment. (Or if your pc is linux or mac, you can use esptool as showed in the attached archive.

When you re-flashed the firmware, see what happened in the UART console ( baud rate is 74880 ). Also you can try to join the SSID of WioLInk with your PC and then try to ping Wio Link, see what happened. If all this failed, please send me piece of debugging messages, and you could fall into the last hope - use the cli tool to setup wio link. 

The cli tool is here https://github.com/Seeed-Studio/wio-cli <https://github.com/Seeed-Studio/wio-cli>, it uses usb cable to config wio link

Hope this will help and look forward to your feedback. 
 
Best,
Jack

----
Seeed Technology Co., Ltd
Software Team

KillingJacky, 2016-03-31:
The upload to github aborts all the time. I put the archive here: http://www.seeedstudio.com/wiki/images/9/92/Firmware_of_wiolink.zip

lawrie, 2016-03-31:
Hi Jack,

I flashed the new firmware.

The instructions in the advanced User Guide are confusing. It says:

a.. Press and hold FUNCTION key, then press RESET key, then release both 

I had to press and hold the reset button, and then the config button and then release both.

I can see I have the new firmware as I do not get the breathing blue LED when I go in to config mode.

I have attached images of the diagnostics I get with baud rate 74880 and 115200. As I said before, the baud rate seems to change to 115200 in config node. I saw a Serial.begin(115200) in the network.cpp code for that path, but was confused about the use of Serial and Serial1 for diagnostics.

At both baud rates, I see nothing when I send the config message from my phone.

Does it make a difference whether I use the USB port for diagnostics, or attach a CP210x device to the TX line on the UART Grove port?  I seem to get the same diagnostics from both.

I will try the cli tool.

Lawrie

From: Jack Shao 
Sent: Thursday, March 31, 2016 3:04 AM
To: Seeed-Studio/Wio_Link 
Cc: Lawrie Griffiths 
Subject: Re: [Seeed-Studio/Wio_Link] Failing to configure (#4)

Hi Lawrie

Sorry for the inconvenience. From the description, I agree with you that the Wio Link doesn’t even recv the config message. I’m sending you a firmware with debugging message enabled, please re-flash the firmware with the guide here: https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link <https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link> the only diff is to use user1.bin & user2.bin in my email attachment. (Or if your pc is linux or mac, you can use esptool as showed in the attached archive.

When you re-flashed the firmware, see what happened in the UART console ( baud rate is 74880 ). Also you can try to join the SSID of WioLInk with your PC and then try to ping Wio Link, see what happened. If all this failed, please send me piece of debugging messages, and you could fall into the last hope - use the cli tool to setup wio link. 

The cli tool is here https://github.com/Seeed-Studio/wio-cli <https://github.com/Seeed-Studio/wio-cli>, it uses usb cable to config wio link

Hope this will help and look forward to your feedback. 

Best,
Jack

----
Seeed Technology Co., Ltd
Software Team

lawrie, 2016-03-31:
Hi Jack,

I successfully used the wio cli tool to setup the device.

Thanks,

Lawrie

From: Lawrie Griffiths 
Sent: Thursday, March 31, 2016 9:08 PM
To: Seeed-Studio/Wio_Link ; Seeed-Studio/Wio_Link 
Subject: Re: [Seeed-Studio/Wio_Link] Failing to configure (#4)

Hi Jack,

I flashed the new firmware.

The instructions in the advanced User Guide are confusing. It says:

a.. Press and hold FUNCTION key, then press RESET key, then release both 

I had to press and hold the reset button, and then the config button and then release both.

I can see I have the new firmware as I do not get the breathing blue LED when I go in to config mode.

I have attached images of the diagnostics I get with baud rate 74880 and 115200. As I said before, the baud rate seems to change to 115200 in config node. I saw a Serial.begin(115200) in the network.cpp code for that path, but was confused about the use of Serial and Serial1 for diagnostics.

At both baud rates, I see nothing when I send the config message from my phone.

Does it make a difference whether I use the USB port for diagnostics, or attach a CP210x device to the TX line on the UART Grove port?  I seem to get the same diagnostics from both.

I will try the cli tool.

Lawrie

From: Jack Shao 
Sent: Thursday, March 31, 2016 3:04 AM
To: Seeed-Studio/Wio_Link 
Cc: Lawrie Griffiths 
Subject: Re: [Seeed-Studio/Wio_Link] Failing to configure (#4)

Hi Lawrie

Sorry for the inconvenience. From the description, I agree with you that the Wio Link doesn’t even recv the config message. I’m sending you a firmware with debugging message enabled, please re-flash the firmware with the guide here: https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link <https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link> the only diff is to use user1.bin & user2.bin in my email attachment. (Or if your pc is linux or mac, you can use esptool as showed in the attached archive.

When you re-flashed the firmware, see what happened in the UART console ( baud rate is 74880 ). Also you can try to join the SSID of WioLInk with your PC and then try to ping Wio Link, see what happened. If all this failed, please send me piece of debugging messages, and you could fall into the last hope - use the cli tool to setup wio link. 

The cli tool is here https://github.com/Seeed-Studio/wio-cli <https://github.com/Seeed-Studio/wio-cli>, it uses usb cable to config wio link

Hope this will help and look forward to your feedback. 

Best,
Jack

----
Seeed Technology Co., Ltd
Software Team

KillingJacky, 2016-04-01:
Hi Lawrie,

The debugging console is UART1, We only expose the TX of UART1 at GPIO2, which is right there beside GPIO13. The release version firmware uses GPIO2 to drive LED, while the debug version uses GPIO2 as UART1 TX.

lawrie, 2016-04-01:
Hi Jack,

But how do I connect to GPIO2? Which pin on which connector is it exposed as? 

Is there a schematic for the board?

Lawrie

From: Jack Shao 
Sent: Friday, April 1, 2016 3:50 AM
To: Seeed-Studio/Wio_Link 
Cc: Lawrie Griffiths 
Subject: Re: [Seeed-Studio/Wio_Link] Failing to configure (#4)

Hi Lawrie,

The debugging console is UART1, We only expose the TX of UART1 at GPIO2, which is right there beside GPIO13. The release version firmware uses GPIO2 to drive LED, while the debug version uses GPIO2 as UART1 TX.

KillingJacky, 2016-04-01:
the silk screen on the board indicates the corresponding GPIO.
[image]

lawrie, 2016-04-01:
Hi Jack,

Thanks for that. I should have realized that they were GPIO numbers.

Here is the diagnostics I get on UART1:

Boot reason: SOFT_RESTART
Node name: esp8266_node
FW version: 1.1
Chip id: 0x008b793b
Free heap size: 25816
Private key: ebd5be8f3451ab74bc070ca25d109e3d
Node SN: 72d0d095dcdd2c357368f07af0aa690a
Data server: 45.79.4.239
OTA server: 45.79.4.239
OTA result flag: 255
enter AP mode ... 
SSID: WioLink_8B793B

----

This confirms that the config message is not being received.

I don’t have a problem any more as I setup the device with wio cli.

Lawrie

From: Jack Shao 
Sent: Friday, April 1, 2016 9:55 AM
To: Seeed-Studio/Wio_Link 
Cc: Lawrie Griffiths 
Subject: Re: [Seeed-Studio/Wio_Link] Failing to configure (#4)

the silk screen on the board indicates the corresponding GPIO.

awong1900, 2016-04-03:
Hi Iawrie,

Still can't configure wiolink with android app? 
I will check this issue with same phone device. 

Ten

lawrie, 2016-04-03:
Hi Ten,

Yes, that’s right.

Out of 50 or more attempts it has only worked once.

It is a Moto G running Android 6.0.

I will try another phone on my network, when I get a chance.

Lawrie

From: Ten Wong 
Sent: Sunday, April 3, 2016 4:42 AM
To: Seeed-Studio/Wio_Link 
Cc: Lawrie Griffiths 
Subject: Re: [Seeed-Studio/Wio_Link] Failing to configure (#4)

Hi Iawrie,

Still can't configure wiolink with android app? 
I will check this issue with same phone device. 

Ten

awong1900, 2016-04-06:
Hi lawrie,

I can't find Moto G with Android 6.0, But I test Moto G with Android 5.0 and Nexus 6 with 6.0, be configure success.   
Please check the version of your app is v2.1.2 ? 

Ten

lawrie, 2016-04-06:
Yes, it is v2.1.2.

Lawrie

From: Ten Wong 
Sent: Wednesday, April 6, 2016 7:19 AM
To: Seeed-Studio/Wio_Link 
Cc: Lawrie Griffiths 
Subject: Re: [Seeed-Studio/Wio_Link] Failing to configure (#4)

Hi lawrie,

I can't find Moto G with Android 6.0, But I test Moto G with Android 5.0 and Nexus 6 with 6.0, be configure success.

Please check the version of your app is v2.1.2 ? 

Ten

retro486, 2016-04-16:
I was having this same issue but I figured out a bit of a workaround that worked on two boards for me: http://blog.rdkl.us/setting-up-the-wio-link/

Basically, make sure you have the correct USB drivers installed for your system, use the wio-cli toolkit, don't wait for the wio to become available, manually enter your wifi details, reset and 4 second config WHILE wio-cli is trying to send the new config to your wio link. Usually after releasing the config button during this step the config goes right in and moves on to testing then you can use the app to configure it as per usual.

KillingJacky, 2016-04-27:
Thanks for your info, Russ.

closed 2016-05-03 by KillingJacky