Seeed-Studio/Wio_Link #5: Phone not connecting

luwii, 2016-03-31:
Received my WIO link two days ago, was super excited, but got stuck very early on. :-(

Using the App V2.1.2 on Andriod V5.0.1 on a Galaxy S4.

After pressing the Config button for 4 seconds, the blue light is mostly (dimly) lit with the odd flash, pressed the 'Got Ready' button and the app should list the WIO to connect to. But nothing happens. Pulling up and down seems to trigger a refresh, but still nothing appears.

I can see the WIO in list of Wifi networks on the phone (if you held down the config button on the WIO). I tried connecting to the WiFi first, as well as during the time I was running the app. The phone connects, WIO gives me an IP, but then I get a 'connect server failure' message in the App when I press the 'Got Ready' button.

Rebooted the phone, 'forgot' the WIO network settings.. 
Followed the instruction to re-flash the WIO, and that process worked, but no change in getting the phone connected. I ran out of ideas to try .... any suggestions?

luwii, 2016-03-31:
I tried the firmware proposed in issue #4 ..  and monitored the serial port at 74880 baud.  The WIO provides the normal (readable) data, then after holding the config button for 4 seconds, I get a long line of gibberish (as if the baud rate is now different)  Pressing buttons on the phone does not add anything to serial output. 

When you re-flashed the firmware, see what happened in the UART console ( baud rate is 74880 ). Also you can try to join the SSID of WioLInk with your PC and then try to ping Wio Link, see what happened. If all this failed, please send me piece of debugging messages, and you could fall into the last hope - use the cli tool to setup wio link.

KillingJacky, 2016-03-31:
Hi luwii,

Which UART port are you connecting to? Pin 2 within the Digital 2 port?

luwii, 2016-04-01:
I was just looking at what was going over the USB cable. Should I connect a 3V3 serial to USB converter cable to one of the other ports?

luwii, 2016-04-01:
Hi KillingJacky, 

I was just looking at the USB port, did not know about Digital port 2. Here is what i get AFTER holding down the config button.

Boot reason: SOFT_RESTART
Node name: esp8266_node
FW version: 1.1
Chip id: 0x000d2f6e
Free heap size: 25816
Private key: ################################
Node SN: ################################
Data server: 255.255.255.255
OTA server: 255.255.255.255
OTA result flag: 255
enter AP mode ... 
SSID: WioLink_0D2F6E

awong1900, 2016-04-03:
Hi luwii,

> I can see the WIO in list of Wifi networks on the phone (if you held down the config button on the WIO). I tried connecting to the WiFi first, as well as during the time I was running the app. The [...]

Reason: The app need get some information from the wio server when press the 'Got Ready' button. But the phone connected on wio AP. So pop 'connect server failure' message.

Back the issue.  If your android version is 6.0. i should suggest you open the location server, cause the new Permission require on Android 6.0 (Scan the wifi list need location permission).

Ten

luwii, 2016-04-04:
Hi Awong,

Thanks for the help! Making sure location services is enabled for the application made it work. Let the fun begin!

closed 2016-04-04 by awong1900