Seeed-Studio/Wio_Link #6: error: time out in notifying the node

yanndouze, 2016-03-31:
Hello

I have this error message when I try to update the node (OTA) : 
> error: time out in notifying the node 

What can I do ?

Best Regards
Yann

KillingJacky, 2016-04-02:
Hi Yann,

KillingJacky, 2016-04-02:
Hi Yann,

This error is basically caused by network latency. Are your Wio Link connecting SEEED's server? Which server are they connecting(international or china)? Which country are you in?

yanndouze, 2016-04-02:
Hi Jack

I'm living in France so I don't really know which server is used ?
Where can I check this information ?
I just take a look at the Wio App and It seems to be the international
server... (http://iot.seeed.cc )

Yann

awong1900, 2016-04-03:
Hi yanndouze,

You can use international sever.

vali77, 2016-04-03:
Hi Yann, i had the same problem today, fix it by moving closer to my router in my case is was a wifi range issue.

yanndouze, 2016-04-08:
Hello

I'm just behind my router and I'm using the international server (Main
server : https://iot.seeed.cc (international) )
What can I do more because I have all the time the same error  ?
Thanks for your help
Best Regards
Yann

awong1900, 2016-04-18:
@yanndouze Can you give me your account? Then i can do more test. 
If can, send it to teng.wang@seeed.cc, include your password and ensure the wiolink is online.

KillingJacky, 2016-04-27:
Close, if further more problems, please reopen.

closed 2016-04-27 by KillingJacky