Seeed-Studio/Wio_Link #8: Connecting to WioLink

zainul1996, 2016-04-08:
Hi there, Ive been trying to connect with WioLink with my phone for the past 10mins yet to no avail. Also what is the IDE we can use to program on it ? Is there any step by step instructions for me to follow as I do not have much experience with iot

awong1900, 2016-04-08:
Hi zainul1996,

Now, you had add a WioLink to your account, or not?

zainul1996, 2016-04-08:
Nope, the process of adding a Wio Link to my account causes the problem. Ive tried both iOS and android. I cannot successfully pair both my boards. I could ender configuring mode but the part that always fails is getting to connect to wifi. The router is next to me so distance cannot be the problem

zainul1996, 2016-04-08:
Error on android :  "Wio Link can not connect to the router. Please check AP password or closer with AP. Please reset Wio to configure mode and try again."  

Error on iOS : "Failed. APConfig: setup canceled or time out!"

zainul1996, 2016-04-08:
@awong1900 Nope, the process of adding a Wio Link to my account causes the problem. Ive tried both iOS and android. I cannot successfully pair both my boards. I could ender configuring mode but the part that always fails is getting to connect to wifi. The router is next to me so distance cannot be the problem.

zainul1996, 2016-04-08:
@awong1900 
Error on android : "Wio Link can not connect to the router. Please check AP password or closer with AP. Please reset Wio to configure mode and try again."

Error on iOS : "Failed. APConfig: setup canceled or time out!"

ErikDeBruijn, 2016-04-09:
Tried for the last hour with two different Wio's. I'm getting the same error, on two different access points, with and without WPA2, with and without 5GHz enabled.

Other devices connect fine.

ErikDeBruijn, 2016-04-09:
I finally got some results. Didn't manage to get the app to configure the device and let it connect using the Android App. However, via USB and my MacBook I could get the device to access my WLAN and access the cloud service. How?

- Open a terminal
- Install the cli tool as follows: `sudo pip install wio-cli`
- Type `wio login` and follow the instructions.
- Run `wio setup`
- Perform the following conversation:
```
> Setup is easy! Let's get started...

! PROTIP: Hold the Configure button ~4s into Configure Mode!
! PROTIP: Please make sure you are connected to the main server

? Would you like continue? [Y/n]: y
> I have detected a Wiolink connected via USB.
> No nearby Wiolink detected.
? Would you like to wait and monitor for Wiolink entering configure mode [Y/n]: n
>
Quit wio setup!
? Would you like to manually enter your Wi-Fi network configuration? [y/N]: y
> Please enter the SSID of your Wi-Fi network: DeBruijnFritzBox
> Please enter your Wi-Fi network password (leave blank for none): bdboksdoorn30
> Please enter the name of a device will be created: WIO_01
> Here's what we're going to send to the Wiolink:

> Wi-Fi network: [ssid here]
> Password: [your pass here]
> Device name: [whatever you like to call it]

? Would you like to continue with the information shown above? [Y/n]: y

> Send Wi-Fi information to device success.
> The Wiolink connect to main server success.
> Set Wiolink name success.

> Configuration complete!
```
From this moment on, the device finally showed up in the list and could be programmed via the Android App.

zainul1996, 2016-04-09:
@ErikDeBruijn I don't have python installed but okay Ill first go thru the hurdle of installing python then pip package manager first 👍

closed 2016-04-27 by KillingJacky

mwcz, 2016-08-06:
THANK YOU @ErikDeBruijn !  I had the exact same problems with the Android app, but the wio-cli package worked great.

laurakurtzberg, 2018-03-27:
Hi! I'm coming to this conversation a couple years later, but I'm wondering if anyone has had trouble with the wio-cli package. Specifically, I get the message 'The Wio Connect to Server Failure' after following all the steps mentioned in the comment above, right after it says 'send wi-fi information to device success'!! Not sure what to try in order to keep troubleshooting.

jonathonadlington, 2019-11-23:
Hi. I keep getting stuck on the "Sending Wi-Fi information to device..." part.  just keeps sending and sending.  Any advice?

awong1900, 2019-11-25:
> Hi. I keep getting stuck on the "Sending Wi-Fi information to device..." part. just keeps sending and sending. Any advice?
Do you also use wio-cli?

jonathonadlington, 2019-11-25:
> > Hi. I keep getting stuck on the "Sending Wi-Fi information to device..." part. just keeps sending and sending. Any advice?
> 
> Do you also use wio-cli?

yeh, I used wio-cli to try and connect to my wifi but it gets stuck right at the end.  the reason I was doing it that way is that when I hold down the config button on my wio link the blue light doesn't come on at all.

awong1900, 2019-11-25:
This can indicate that the hardware or firmware itself is damaged. Is there another wiolink or wionode that can be cross-verified?

jonathonadlington, 2019-11-25:
> This can indicate that the hardware or firmware itself is damaged. Is there another wiolink or wionode that can be cross-verified?

No I haven't got another one.