Seeed-Studio/Wio_Link #10: Android app has trouble connecting to Wio Link

jarkman, 2016-04-11:
I've just received my Wio Link, and I'm trying to connect from a Nexus 6P.

I have two problems:
(1) The initial connection only works very rarely, perhaps 1 time in 10. I see the Select Wio Link form, tap on my Wio Link, then see 'Connecting to WioLink_xxx' and a spinner on a popup. That popup rarely goes away. 

(2) Occasionally, if I tap Back and select the WioLink_xxx again, I see the Select AP form. I can pick an AP, then I see 'Sending wifi password' and 'Waiting Wio Link get ip address'. However, that always ends with this:

"Wio Link cannot connect to the router. Please check AP password or closer with AP. Please reset Wio to configure mode and try again".

It looks to me as if the communication between Android and Wio Link is just very flaky.

What should I try next ? 

Thanks,

Richard

awong1900, 2016-04-11:
Hi jarkman,

1) I think that the application may be have compatibility problems can not all work well on all platforms. Especially automatic switching network.

2)General error is that Ap password is not correct.

But If you are difficult configure with app.  I suggest you use Wio Link Command Line Toolset.
https://github.com/Seeed-Studio/wio-cli

first login, then use
$wio setup

Sorry for inconvenient

jarkman, 2016-04-11:
Hi - thanks for the quick reply.

I have installed wio-cli, and it does find the Wio Link.

But, I've tried the details for two different APs, and both times I get this:

? Would you like to continue with the information shown above? [Y/n]: 

> Send Wi-Fi information to device success.                                      

[timer counts down from 30 to 0]

>> The Wiolink connect to main server failure.                                   

> Please check info you enter, Try again.

Are there any constraints on the AP that I should know about ? Band, or auth type, or anything like that ?

Thanks,
Richard

awong1900, 2016-04-11:
Hi jarkman,

> Are there any constraints on the AP that I should know about ? Band, or auth type, or anything like that ?
Only thing is the SSID name must be English characters or number. Any auth type even without a password.

Ten

awong1900, 2016-04-11:
Hi jarkman,

Last method to fixed it. Refer to https://github.com/Seeed-Studio/Wio_Link/issues/4, updated the firmware.

zainul1996, 2016-04-17:
@awong1900 hey there, I tried the cli and kept getting "No found device! Plese connect your Wiolink with USB.  "

zainul1996, 2016-04-17:
@awong1900 Hey I keep getting this, Hope someone can help me fix this problem
"Would you like to wait and monitor for Wiolink entering configure mode [Y/n]: y"
"Waiting for a wild Wiolink to appear... (press ctrl + C to exit) "

awong1900, 2016-04-18:
@zainul1996  you need press the Configure Botton ~4s, then wio-cli will found the wild Wio Link.

zainul1996, 2016-04-18:
@awong1900 But now I yet face another problem which is that I cant seem to scan for nearby Wi-Fi network. All it shows is "Asking the Wiolink to scan for nearby Wi-Fi networks... “. I tried entering the SSID and Password manually but it just timeouts. Hope to hear from you soon :) Thanks for all ur help btw

awong1900, 2016-04-19:
@zainul1996  I think the firmware is not latest by your description.
Can you refer to https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link manually update the firmware.

zainul1996, 2016-04-19:
@awong1900 awong I have issues with doing that as well. I have linked below

[image]

[image]

awong1900, 2016-04-19:
@zainul1996 your pc have installed the usb driver or not? 
download url:
https://www.silabs.com/products/mcu/Pages/USBtoUARTBridgeVCPDrivers.aspx

zainul1996, 2016-04-19:
yeapp I installed it before starting the whole process :(

zainul1996, 2016-04-19:
@awong1900 yeapp I installed it before starting the whole process :(

awong1900, 2016-04-19:
@zainul1996 If I unplug the usb cable, your errors will be reported, if everything is correct, it shows
[image]

Check your ports, if the com connected, it will show here.
[image]

zainul1996, 2016-04-19:
@awong1900 This is all I have :/
[image]

awong1900, 2016-04-19:
@zainul1996 Can you give me you KS account name. If we think it is a hardware problem, we will re-send you a new one.

awong1900, 2016-04-19:
@zainul1996 
The serial numbers do not match. 
[image]

closed 2016-04-27 by KillingJacky

gautampoornachandra, 2016-11-04:
I am trying to configure the Wio Node from the Wio app using my Nexus 5 with Android 6.0.1.

Intially issue was solved when the location services was enabled.

After that the, I got the message as Sending Wi-Fi password to Wio. After that I got the following message:

Connection Error
Your phone disconnected from Wio's Wi-Fi hotspot.
Error code:1042

How to connect using Android Phone without using a client?

awong1900, 2016-11-05:
@gautampoornachandra The code 1042 mean "your phone disconnect from WIO hotspot".
The reason maybe the app change WIFI network failure automatically. You can check the System wifi setting, ensure phone connect WIO hotspot, named "Wio_xxx". If not, you can manually change to Wio hotspot, then back to step two, go on...

jamesdinsmore, 2017-05-21:
I have same problem as above:
I am trying to configure the Wio Node from the Wio app using my Nexus 5 with Android 6.0.1.

Intially issue was solved when the location services was enabled.

After that the, I got the message as Sending Wi-Fi password to Wio. After that I got the following message:

Connection Error
Your phone disconnected from Wio's Wi-Fi hotspot.
Error code:1042
My phone was connected to the Wio; but it gets kicked off before the Wio connects to my home wifi network.

awong1900, 2017-05-22:
@jamesdinsmore 
It has some chance to fail, app send wifi configuration information via UDP to wio device. Then the wio device tries to connect ap. Please try it several times. Nexus 5, i have tested it. If still not, try replacing an AP.  

And we have alternative solutions is Python CLI tools, it also help to setup wio device. 
https://github.com/Seeed-Studio/wio-cli