Seeed-Studio/Wio_Link #9: build_firmware.py fails

Thorvin, 2016-04-11:
Hi,
I'am having trouble building the firmware:

```
user_code_pack.a(core_esp8266_main.cpp.o): In function `init_done()':
core_esp8266_main.cpp:(.text+0x54): undefined reference to `cont_run'
user_code_pack.a(core_esp8266_main.cpp.o): In function `loop_task(ETSEventTag*)':
core_esp8266_main.cpp:(.text+0x7b): undefined reference to `cont_run'
user_code_pack.a(core_esp8266_main.cpp.o): In function `do_global_ctors()':
core_esp8266_main.cpp:(.text+0x128): undefined reference to `cont_yield'
core_esp8266_main.cpp:(.text+0x142): undefined reference to `cont_yield'
collect2: error: ld returned 1 exit status
make: *** [user1.elf] Fehler 1
core_esp8266_main.cpp:(.text+0x54): undefined reference to `cont_run'
```
Some research led to a missing 'cont.S' assembler module but I have no clue how to proceed :(

Any hints?

Regards
Thomas

KillingJacky, 2016-04-11:
Thanks for reporting. cont.s is back.

Thorvin, 2016-04-11:
Hi,
Thanks for the fast reply! It works :)

Thomas

closed 2016-04-17 by Thorvin