Seeed-Studio/Wio_Link #11: debugging messages

Thorvin, 2016-04-12:
Hi,

I have my WioLink connected via USB and can see the startup messages in a terminal on ttyUSB0. Setting 
`ENABLE_DEBUG_ON_UART1 1` 
in ./node_main/user_config.h the blue led stays dark and i expected to see the output from Serial1.println() statements also - but nothing shows up. Did i miss some further steps to enable debugging? 

Regards
Thomas

KillingJacky, 2016-04-13:
Hi,

Please check the following:
1) OTA at least once after setting ENABLE_DEBUG_ON_UART1 1 
2) Debugging UART's tx pin is GPIO2
3) Band rate of debugging serial is 74880

ps: blue led staying dark is right, because gpio2 is used as tx pin of uart1 now, not to drive the blue led.

Thorvin, 2016-04-13:
Hi,
Yes to all, supposing the blue led staying dark proves 1) and 2). Baud is correct as i can see the startup just fine:
```
--- Miniterm on /dev/ttyUSB1  74880,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+B | Help: Ctrl+B followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

��+P�*��h�*�Dr 
�
```
... except for the garbage at the end.  
I found the following in ./node_main/network.cpp:
```
#if ENABLE_DEBUG_ON_UART1
    Serial1.begin(74880);
    //Serial1.setDebugOutput(true);
    Serial1.println("\n\n"); 
#else
    pinMode(STATUS_LED, OUTPUT);
    digitalWrite(STATUS_LED, 1);
#endif
```

I tried changing the 'greeting line there, installed and tried the grove_example module (which has several serial outputs) and tried to add Serial1.println() to grove_relay but no output in the terminal :(

Regards
Thomas

KillingJacky, 2016-04-13:
Hi Thomas,

I still want to check with you the 2) point. I mean you should connect your usb to serial convertor's RX pin with GPIO "2" of Wio Link. GPIO "2"  is the 2nd pin on the left of "Digital2" port, not the "UART" port. The words included in "" is the silk screen.

More info... both uart0 and uart1 will print booting message at the beginning of bootup. So you get piece of booting message won't prove you wire the lines correctly.

KillingJacky, 2016-04-13:
[image]

Thorvin, 2016-04-13:
Ahhhh, so sorry, seems that this was a complete misunderstanding from my side :(
Currently, I just have the WioLink's micro usb  connected to my PC and as the bootup messages showed up in the resulting ttyUSB I thought i could get debugging show up there too...

Ok, so I'll have to dig out my converter and build the cable...

Thomas

Thorvin, 2016-04-13:
Hi,
It works :) Many thanks for your patience and sorry for the noise...
Thomas

KillingJacky, 2016-04-13:
You're welcome

closed 2016-04-17 by Thorvin