Seeed-Studio/Wio_Link #12: grove_ir_receiver compatibility

Thorvin, 2016-04-14:
Hi,
Combining  grove_ir_receiver with grove_temp_hum, reading the sensor crashes my Wio in ~2 out of 5 calls. I could trace the crash to `suli_delay_ms(250);` in `GroveTempHum::_read()`. I did some more tests and it seems that **any** call to `suli_delay_ms()` causes a restart (sooner or later) when the IR Receiver is active.

Debugging console shows:
```
Boot reason: EXCEPTION
Node name: esp8266_node
FW version: 1.1
Chip id: 0x008b3609
Free heap size: 25072
```
In the Terminal connected to the micro-usb:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user2 @ 101000

don't use rtc mem data
rl��rl��
```

Could this be a general problem caused by the 50 µs timer interval in grove_ir_receiver?

Thomas

KillingJacky, 2016-04-18:
Thanks, I will look into this.

KillingJacky, 2016-04-19:
Hi Thomas,
It turns out to be a bug of ir receiver driver, and get fixed. Please OTA your wio link once again. Hope the fix works for you. And thanks again for reporting the issue.

Thorvin, 2016-04-20:
Hi,
Many thanks, looks good so far! 

Thomas

P.S.  (Unrelated): The latest Version of the Android App shows a new "Wio Node" module when adding a device but I couldn't find it mentioned on the Seeed-Homepage? Just curious...

KillingJacky, 2016-04-21:
You're welcome. 

Wio Node does exist, under manufacture. Will see it soon :D

closed 2016-04-27 by KillingJacky