Seeed-Studio/Wio_Link #14: On Ubuntu 14.04 32-bit driver database won't build, no errors given.

retro486, 2016-04-16:
Following the instructions here: https://github.com/Seeed-Studio/Wio_Link/wiki/Server%20Deployment%20Guide#2-deploy-directly-into-host-filesystem

I'm using the 32-bit linux precompiled toolchain from http://arduino.esp8266.com/package_esp8266com_index.json on a digital ocean Ubuntu 14.04 droplet.

I'm able to get the test server running just fine, and the scan_drivers.py script spits out a few "deprecated conversion from string constant to 'char*'' warnings, but no errors.

At the end, driver_docs.json is an empty array, drivers.json is an empty array, and scan_status.json contains:

`{"result":"OK", "msg":"scanned 0 grove drivers at 2016-04-16 20:00:55.885686"}`

So of course build_firmware.py fails with "can not find GroveRelay in database".

Looking at scan_drivers.py, it appears that the only time driver.json would be empty is if there's an error, which would be logged in scan_status.json, but that isn't happening, so I'm not sure what's up...

Thorvin, 2016-04-17:
Hi,

Are you sure your `grove_drivers/` isn't just empty? IIRC i had a similar problem and finally downloaded and installed the drivers manually in this directory. 

Thomas

retro486, 2016-04-17:
Ah that's exactly it. I didn't see that folder is a link to another repo. Thanks! Also, after cloning the grove drivers github there I had to remove the .git folder since scan_drivers.py apparently tries to pick it up and errors out. But fixed! Firmware builds perfectly.

closed 2016-04-17 by retro486

KillingJacky, 2016-04-18:
Thanks for reporting, the guide needs to be updated too. We will fix this slip.