Seeed-Studio/Wio_Link #15: Build dependencies not documented

9037568, 2016-04-16:
Build dependencies are not documented, but should be.

When attempting to compile the project, I get this error:

>X:\projects>build_firmware.py
Traceback (most recent call last):
  File "X:\projects\build_firmware.py", line 27, in <module>
    import yaml
ImportError: No module named yaml

Further, if I try to install yaml, I get this error:

>X:\projects>pip install yaml
Collecting yaml
  Could not find a version that satisfies the requirement yaml (from versions: )
No matching distribution found for yaml

KillingJacky, 2016-04-18:
Hi, is this what you concerned? https://github.com/Seeed-Studio/Wio_Link/wiki/Server-Deployment-Guide#22-python-modules-required  Did you deploy your private server with docker?

9037568, 2016-04-18:
Well, that explains why I didn't find that information. I'm not deploying to a private server, just trying to compile the project. Thanks for the pointer.

closed 2016-04-18 by 9037568