Seeed-Studio/Wio_Link #17: Update firmware

zylopezr, 2016-04-28:
Hello,  I've worked with an conventional esp8266, but I've bougth your target and I'd like to update a firmware from my computer, could you tell me how to connect the pins?  (I have a usb-ttl convertor)

KillingJacky, 2016-04-29:
Hi, please download the schematic of wio link here http://www.seeedstudio.com/wiki/File:Wio_Link_v1.0_sch.pdf. Then you can refer to the wire connection of wio link to connect your esp8266 board together.

closed 2016-05-03 by KillingJacky

zylopezr, 2016-05-03:
Hello, me again. 
I've been trying to update the firmware using Ubuntu, I'm following this tutorial,  https://learn.adafruit.com/downloads/pdf/building-and-running-micropython-on-the-esp8266.pdf
I have to connect the board to usb-ttl convertor, but I've not found the correct way, Advanced User Guide says one have to connect UART1 TX to the pin RX of convertor and also the Ground line, I did it but I got an error "Inappropriate ioctl for device" like if the board is not well connected. Do you have any information about updating firmware using Ubuntu, or about this error?
Thank u so much.

moxondesign, 2016-08-09:
Hi, That link is broken, where is the schematic please?

moxondesign, 2016-08-09:
found it here - http://www.seeedstudio.com/wiki/File:Wio_Link_SCH_v1.0.rar