Seeed-Studio/Wio_Link #19: Update firmware using ubuntu

zylopezr, 2016-05-03:
Hello, me again. 
I've been trying to update the firmware using Ubuntu, I'm following this tutorial, https://learn.adafruit.com/downloads/pdf/building-and-running-micropython-on-the-esp8266.pdf
I have to connect the board to usb-ttl convertor, but I've not found the correct way, Advanced User Guide says one have to connect UART1 TX to the pin RX of convertor and also the Ground line, I did it but I got an error "Inappropriate ioctl for device" like if the board is not well connected. Do you have any information about updating firmware using Ubuntu, or about this error?
Thank u so much.

KillingJacky, 2016-05-04:
Hi, using Linux you can try this tool https://github.com/igrr/esptool-ck

example command I'm using: esptool -cp /dev/tty.SLAB_USBtoUART -cd nodemcu -cb 921600 -ca 0x1000 -cf user1.bin -ca 0x101000 -cf user2.bin

KillingJacky, 2016-05-04:
The wiring is simple, just connect Wio Link to PC with micro usb cable (you need installing linux driver for CP2102). Find the correct device descriptor for CP2102 under /dev/, for me it's /dev/tty.SLAB_USBtoUART. The esptool will help handling initialization before downloading firmware.

zylopezr, 2016-05-04:
I'm trying with this code
python esptool.py -p /dev/ttyUSB0 --baud 74880 write_flash --flash_size=8m 0 firmware-combined.bin
because I need to run Micropython on the Esp, I've connected the board like you said, but the error persist.

KillingJacky, 2016-05-04:
esptool mentioned by me is not a python script. esptool.py is another project.

awong1900, 2016-07-16:
@zylopezr I test this cmd, it work for wio link.
```
esptool.py -p /dev/cu.SLAB_USBtoUART -b 921600 write_flash -fs 32m-c1 -fm qio --verify 0x000 bootloader.bin 0x00001000 user1.bin 0x00101000 user2.bin 0x003fc000 esp_init_data_default.bin 0x003fe000 blank.bin
```

KillingJacky, 2016-07-27:
As long time no response, we assume that you've passed this issue. We close this issue for now.

closed 2016-07-27 by KillingJacky