Seeed-Studio/Wio_Link #22: Connecting to WioLink_XXXXXX

jaime-axt, 2016-05-11:
Hi guys! I am having some troubles here.

- I downloaded the app (for Android) and I created my account
- I connected my phone to my WIFI network and I enabled the GPS
- I connected the WIO Link to the power and press the config button (for 4 seconds) 
- I press the "GOT READY" button and I can see the WIFI of the WIO LINK (WioLink_XXXXXX)
- Whe I press WioLink_XXXXXX, the phone loss the WIFI connection and the app show "Connecting to WioLink_XXXXXX"... and I am still waiting...

My phone is an Android 5.0.1 (Galaxy S4)

I thought that WIO LINK was bricked, so I tried to follow the procedure indicated in the Advanced User Guide, but I receive another message: COM FAIL (serial port open error)

[image]

Please help!!!

awong1900, 2016-05-12:
@jaime-axt 1. Ensure COM PORT is right.  See 
[image]
2.  If the app stuck in "Connecting to WioLink_XXXXXX"... ,  try to manually connect to WioLink_xxx on the system WI-FI settings.

jaime-axt, 2016-05-12:
I configured manually the wifi to connect to WioLink, and now works perfect.

Thank you very much!

closed 2016-05-13 by awong1900