Seeed-Studio/Wio_Link #23: configure server for port other than 8080

tube0013, 2016-05-12:
The box I want to run the server is on, already has a service using 8080, how can I have this (with docker image) run on a different port?

thanks.

KillingJacky, 2016-05-13:
Get access to your docker container, edit /root/esp8266_iot_node/server.py, find `http_server.listen(8080)` , change 8080 to your available port. Stop the container. Commit your container. Instance another container with the committed one.
```shell
sudo docker ps 
sudo docker commit <containerid> <foo/live>
sudo docker run -itd  --name p2 --restart=always -p 127.0.0.1:xxxx:xxxx -p 8081:8081 -p 8000:8000 -p 8001:8001 -v /root/esp8266_iot_node <foo/live>
```

tube0013, 2016-05-13:
thank you.

I got the docker image up and running on an accessible port.  I can post commands with curl and get responses, including creating a user, and logging in.

with the iOS app, i've changed to a custom server, and attempted to re-configure my wio link.  I've gotten it to be discoverable, and go through the process to setup the network, which is successful (I see it on my wifi network and can ping it).  It fails to connect to my server though.  I'm not sure why as my iphone and computer with wio-cli are on the same network and can access the server fine.

tube0013, 2016-05-13:
disregard, I'm working on this remotely from work, and port 8000 and 8001 needed to be opened to get to the server remotely.

tube0013, 2016-05-13:
Thank you.  The server is up and running.  iOS app is configured to connect to it successfully, and I can create a user, and add my wio link device.  

if I logout using the app and attempt to log back in, the app crashes.  the only way to get back in is to create a new user account.  (the api does not seem to have a way to delete users?)

Once in, I can see my wio link online, and I try to set up my groove sensors, and update the firmware.  the wio link goes into ota mode (blue led blinks fast), and phone says "downloading firmware" but it eventually fails.

looking in the docker /root/esp8266_iot_node/users_build/xxxx/  the build log says the firmware was successfully built.  so it appears the issue is the firm ware going to the wio link.  

I am working remotely from the server, with the wio link local to me but have opened all the ports (8000,8001) on the remote server.

As everything seems to work (besides the app crash) I'm at a loss as to why the firmware that is built is not downloading to the wio link.

KillingJacky, 2016-05-16:
Hi, 
Thank you for providing the crashing informations. Would you please tell me the version of your iOS app? ( to get the version, swap from the left edge of the main UI as far as possible to the right edge, the version information will be at the footer of the lower layer). I will forward the crashing information to our app developers.

Regarding the failure of downloading the firmware, did you ever change the port 8081? because Wio will engage a http request to the 8081 port on the server to download the firmware binary. For convenience of your debugging, I would like to introduce the use of all the ports that server listens on.

8080 - a http server to expose http APIs for Grove property read / write
8081 - a http server which serves the firmware downloading
8000 & 8001 - tcp server to be connected by wio link

To change 8081/8000/8001 is more complicated than changing 8080 because it will involve modifications for firmware.

Another tip for the cause of downloading failure is that big network latency will also lead to the failure. What’s the ping latency and packet loss rate from your local to your remote server?  

----
Jack Shao
邵旭光
Software Team
Seeed Technology Co., Ltd

tube0013, 2016-05-16:
App version is 1.4.2(599)

I figured out the ports, and was my port 8081, once I opened that up on my router the firmware download succeeded. 

I have the wio now set up and working.  With home-assistant.io pulling in data via curl.

KillingJacky, 2016-05-16:
Thank you and glad you get your Wios working ;)

closed 2016-05-25 by tube0013