Seeed-Studio/Wio_Link #27: API variables changing

6A, 2016-07-19:
I have a custom program, which declares a boolean variable named **strict**.

Using the API, I can change the variable without a problem. But it looks like it sometimes resets.

`curl https://us.wio.seeed.io/v1/node/variable/strict` -> "0"
`curl -X POST https://us.wio.seeed.io/v1/node/variable/strict/1`
`curl https://us.wio.seeed.io/v1/node/variable/strict` -> "1"

I'll check it again a few hours later:
`curl https://us.wio.seeed.io/v1/node/variable/strict` -> "0"

The variable can only be change throu the Internet, and this behavior appeared hours after pushing the update, so I'm 100% sure that nobody's playing with my API tokens.

I guess the Wio Link restarts every few minutes / hours, since "0" is the default value of my variable. If so, where is it documented?

KillingJacky, 2016-07-20:
Hi, thanks for reporting this issue. We will test it using your scenario, and to see if the issue is  recurrent. For now, there're 2 possible reasons I can imagine that the variable resets. 
- exception happen, and caused the wio to restart
- network connection broken or slow, heartbeat lost for a long time (1min), wio actively reboots itself. 
Anyway, we will look into this and if you have any progress please let us know too.

KillingJacky, 2016-07-20:
BTW, from the URL, I guess you've uploaded your custom program to the newest global server, is that true? Or which firmware version are u using if you deployed a private server?

6A, 2016-07-20:
@KillingJacky Hi, thanks for your answer!

Nope, I'm using the new global server. The project can be found [here](https://github.com/6A/wio-my-light).

I didn't think about the possibility of an exception, although the resets seem "random", so I don't really know what would cause it.

KillingJacky, 2016-07-20:
@6A Nice project and docs.  I already cloned your project with my wio link. Instead my wio link connects to my localhost private server. Let's see how is it going for mine copy :)

6A, 2016-07-20:
@KillingJacky Thanks! Lemme know how it goes!
I didn't include a picture of where to plug what though, so I hope it wasn't too hard to find it out!

KillingJacky, 2016-07-20:
It's not hard for me to figure out which grove connects to which port. But for others I'm afraid it will take a while. Well here's the result at my side. It does reboot for no reason. I enabled the debug printing, and here's what I get:

```shell
Node name: Wio Link 22
FW version: 1.2
Chip id: 0x0015ec67
Free heap size: 24200
Private key: 7e5188c22b63fa67acde18f13b3969bf
Node SN: f6bca7d7ef9bb00eb48264b15b55d051
Data server: 192.168.31.12
OTA server: 192.168.31.12
OTA result flag: 0
start to establish network connection.
connect to ssid Xiaomi_Youth with passwd depot0510se
Waiting ip, state: CONNECTING
err: wakeup failed!
Waiting ip, state: CONNECTING
wakeuped
paj7620 initialization done.
Waiting ip, state: CONNECTING
Done. IP: 192.168.31.236
Resolving 192.168.31.12...
Resolved IP: 192.168.31.12
Resolving 192.168.31.12...
Resolved IP: 192.168.31.12
data connection init.
ota connection init.
data conn connected
ota conn connected
data conn confirm hello: 1 
data conn handshake done, keep-alive
ota conn confirm hello: 1 
ota conn handshake done, keep-alive
ota conn: ##PING##
ota conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
data conn: GET /variable/strict

data conn: GET /variable/strict

ota conn: ##PING##
data conn: POST /variable/strict/1

data conn: GET /variable/strict

ota conn: ##PING##
data conn: ##PING##
data conn: GET /function/toggle

ota conn: ##PING##
data conn: POST /function/toggle/bm9uZQ==

data conn: POST /function/toggle/bm9uZQ==

data conn: GET /variable/strict

data conn: GET /variable/strict

data conn: GET /variable/strict

data conn: GET /variable/strict
[... 205 lines cut ...]
data conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
ota conn: ##PING##
data conn: ##PING##
.....
```

From the boot reason (DEFAULT) I can see the boot is just like a power-on startup. It's not caused by exception either watchdog. And the time point it reboots is totally random too. For the last try, it keeps running good for 2 hours, but the worst one is 10 minutes (even when I did not issue any internet request). 

It's just accidental and not regularly recurrent. I have to say it's a tough issue. What's in my brain now is that maybe the LED-Strip causes the power unstable, so the ESP8266 is just like going into a power down and up. I'm going to do more experiments - remove the LED-Strip...etc.

6A, 2016-07-20:
@KillingJacky Wow, thanks for the very thoughtful reply! Btw, I know this is a private server, but do watch out since you included a password / token in your logs!

Now the issue... Well it does look weird, thanks for investigating!

KillingJacky, 2016-07-25:
After twice reboot, it never happen again. The Wio can stand very very long time without reboot. I even didn't change anything of the environment. It's really odd.

KillingJacky, 2016-07-25:
How many LEDs are in your strip? 60? I still think the power drain of the LED strip caused the unstable state.

6A, 2016-07-25:
It says there are 60 of 'em, but as you see in the code I only turn on 30 LEDs, and that lights up the whole strip. I'll try, thanks! Btw, I turn off the strip by setting everything to black (`000000`), is this the correct way to go?

KillingJacky, 2016-07-27:
I think you just pull the strip off and test if it still reboots. Or you try to pick the red wire out of the strip's grove socket and connect it ( vcc of led strip ) to a 5V power.

6A, 2016-07-27:
Damn, I'll do that some other time, right now everything is already set up and doing this'll take hours. Wanna close this? I'll try it out when I get the opportunity, and reopen this issue if needed?

KillingJacky, 2016-07-27:
Sorry man, I really can't get it recurrent afterwards. Thank you for your understanding and your possible future efforts.

KillingJacky, 2017-07-19:
Will close this, feel free to reopen it if any further question.

closed 2017-07-19 by KillingJacky