Seeed-Studio/Wio_Link #28: Syncing Data....

Redrooster1515, 2016-07-23:
Hello I just received my Wio Link and Wio Node. I have tried to follow the instructions and find that on Step 3 the screen indicates"Syncing Data"...**We are almost there! This process will take about a minute**. This message with the spinning circle remains without finishing. Am I doing something wrong? I have tried all the servers available with the same result, and both the Node and the Link. The instructions on the website as well as the videos on YouTube are difficult to follow  as the app on my Android phone does not look like the examples. Due to the app being more recent on my phone? It would be nice if the site was updated showing the most recent app.
Any advise would be helpful .
Thank you.
Rob./

awong1900, 2016-07-23:
I think the firmware of your device is too old and is not be OTA once. It's my android app mistake, and i will fix it.
Now there is an alternative solution.
You can try use CLI to add device.https://github.com/Seeed-Studio/wio-cli
Or you can re-flash the firmware, refer to https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link

Redrooster1515, 2016-07-24:
Thanks for the quick reply. Well I must say i am not happy that i have to upgrade my firmware on a unit that i recieved 2 days ago? Updating the firmware is not easy . The entire idea of purchasing this unit is the simplicity of use."3 steps..5 minutes". If I have to flash ...which I have tried using the Tool. ( I get "chip sync error" .....and with the PIP tool i go as far as getting the error Type Error TypeError: a bytes-like object is required, not 'str'. I have already spent 4 hours trying to connect to either my Node or the Link mostly because the website instructions contain conflicting info or insufficient instructions.
see below....
\Getting device information...    load_entry_point('wio-cli==0.2.2', 'console_scripts', 'wio')()
                                                     File "c:\python36\lib\site-packages\click\core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "c:\python36\lib\site-packages\click\core.py", line 696, in main
    rv = self.invoke(ctx)
  File "c:\python36\lib\site-packages\click\core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python36\lib\site-packages\click\core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python36\lib\site-packages\click\core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "c:\python36\lib\site-packages\click\decorators.py", line 64, in new_func
    return ctx.invoke(f, obj, *args[1:], **kwargs)
  File "c:\python36\lib\site-packages\click\core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "c:\python36\lib\site-packages\wio\commands\cmd_setup.py", line 497, in cli
    r = serial_send(msvr, msvr_ip, xsvr, xsvr_ip, node_sn, node_key, port)
  File "c:\python36\lib\site-packages\wio\commands\cmd_setup.py", line 228, in serial_send
    if 'Node' in ser.readline():
TypeError: a bytes-like object is required, not 'str'

Regards,
Rob

eladpaz, 2016-07-24:
Please Help! I bought the wio link and wio node, and am experiencing exactly what Rob wrote.
I am not happy!!

kindly assist,
Elad

awong1900, 2016-07-25:
@Redrooster1515 wio-cli just support python2. 
On Python3, use virtualenv to install:

    $ virtualenv -p python2 env2
    $ source env2/bin/activate
    $ pip install wio-cli
    $ wio --version

Sorry for the inconvenience

Stevegeocacher, 2016-08-02:
I tried the new update android app..... still cannot connect.

Redrooster1515, 2016-08-02:
Steve. I have tried all methods given to update the firmware..no luck. It comes down to poor instructions. I always run into a dead end. The developers of the product have let us down. The boards should have been sent out with the proper firmware. The whole intent of this product is to be up and running in 5 minutes.... I have spent in excess of 5 hours now and I give up trying to update the firmware. The information that I have tried does not work...no clear instructions. I have bought both the Wio Node Wio link and a bunch of Grove sensors ...all useless to me now. I now know not to support this company and have moved on to the Arduno mega that has lots of support. Wio is a complete marketing failure in my opinion. Cheers.

Stevegeocacher, 2016-08-02:
I was trying to connect my original kickstarter wio link board (from the kickstarter fundraiser), using the android app. It does NOT look like any of the instructions. I just received an email, stating that there is a new updated android app that will connect.  Hmm..... I have V2.3.1 currently. Is that the new one? If so, it doesn't work either.

KillingJacky, 2016-08-02:
@Redrooster1515 @Stevegeocacher @eladpaz Sorry our guy didn't give you the direct guide to fix your problem. Actually reflashing the firmware is the last last last choice to fix problems, because it's really not an easy thing for most peoples. There was a bug in our Android App and we released a bug fix version(2.3.1). Now you gave us that the version 2.3.1 still not working, that's really a bad user experience again. Sorry for that. But we're still trying to figure out the reason, if you would like to send back some information of the failure scene, that will be much appreciated. Here's a check list,
- What's the status of the blue LED when the Wio setup process gets stuck? (The blink patterns of LED are described in App -> Support) 
- When did you purchase the Wio devices? 
- What's the model of your cell phone?

eladpaz, 2016-08-02:
@KillingJacky , thank you. I eventually managed to set up the wio link, but my computer (windows 10) still doesn't recognize the wio node at all. when I connect the node, there is no sign of it on the computer (no onnection sound, no port open, nothing...), however it does give power to the node. I would appreciate your assistance. 
thank you,
elad

Stevegeocacher, 2016-08-03:
I've been able to finally connect for a minute..... Now, when I try to update the wiolink to connect the accelerometer in the I2C spot, I get errors.  I'll try again later. I now need to charge up my phone.

KillingJacky, 2016-08-03:
@eladpaz One of the differences between Link and Node is, Node has no USB to Serial converter chip (CP2102 at Wio Link). If you want to reflash Wio Node's firmware, you've got to find a converter and connect Wio Node's UART port. (still want to tip you, reflashing firmware is the last solution you should take) If you were trying to use wio-cli tool to configure the Wio Node, you have another alternative choice - the wireless way. wio-cli tool can also be working via wireless.
BTW, how did you get your Wio Link work eventually? Reflashing the firmware?

Stevegeocacher, 2016-08-03:
https://youtu.be/64o7J0QUqpo

video of me trying to link up with wiolink. After the 4th try, it just worked.  Mine was one of the original kickstarter kits.

Redrooster1515, 2016-08-03:
Yes exactly the same problem here....Many thanks for the video. I don't know if I have the patience to try dozens of times though. Lol
Cheers

KillingJacky, 2016-08-03:
@Stevegeocacher Thank you very much for your video, that's helpful. Our App developer is investigating now. It seems that this problem only happens on Android phone. We're studying what's the difference in the configuration logic between iOS and Android App.

KillingJacky, 2016-08-03:
@Redrooster1515 @Stevegeocacher Based on the error message from Steve's video, the App developer found out that the configuration process fails on the WiFi AP switch. On some models of Android phone, the system refuses to switch the WiFi AP connection too frequently (connect to router -> wio link -> router again). We'll roll out a hotfix version ASAP. Thank you guys very much for helping us improve!

Redrooster1515, 2016-08-15:
Hotfix....?

KillingJacky, 2016-08-15:
New version has been pushed to google app store but still under verification. You can download it from this test site https://www.pgyer.com/wiolink

sniffedagain, 2016-09-11:
@KillingJacky this product is plagued with issues. After going through all of your steps, flashing firmware, updating to the unapproved WIO LINK software your WIOLINK board refuses to sync / receive password from your app. This product is essentially useless on its on. I am using nougat. Here is a caveat to developers of your ilk: This product  was supposed to be intended for my kids. Because you rushed to market have absolutely horrible quality control and obviously don't bother with best practices there are 100s if not 1000s of these boards out there doing nothing. You should be ashamed of yourself and get this right ASAP.

KillingJacky, 2016-09-11:
@sniffedagain Are u using android phone? Which model? AFAIK, the issues mostly happen for android phone, because there're so many brands and models of them, and the OS installed are multifarious too, the access privilege for network/WiFi is really hard to be acquired for all the cases. The setting up works fine on iOS though. I'm sorry for bring you so much anger. Taking all of your blames, we will thank you in advance if you can provide more details about the failure,  e.g. a video or descriptions, for us to debug the App.

befreestudios, 2017-05-05:
Just got two kits from MassDrop and having the same issues as stated here. Right out of the box.

```
Connection Error
Your phone disconnected from Wio's Wi-Fi hotspot
Error code: 1042
```

awong1900, 2017-05-09:
Hi @befreestudios  Need some patience here. There are some instability in different Android phones and routers. You can try change router or use IOS app to try again.  We have Wio CLI tools as the last option .

KillingJacky, 2017-07-19:
Will close this, feel free to reopen it if any further question.

closed 2017-07-19 by KillingJacky