Seeed-Studio/Wio_Link #29: 400 Client Error: Bad Request for url: https://iot.seeed.cc

airdissy, 2016-07-25:
Hello, 

I'm trying to create my own server to test grove drivers for wio link.
I installed python and all necessary packages but I'm still blocked here
`[I 160725 22:14:02 server_lean:824] Fetching the list of nodes...
`
`[I 160725 22:14:02 connectionpool:788] Starting new HTTPS connection (1): iot.seeed.cc
`
`[E 160725 22:14:03 server_lean:837] 400 Client Error: Bad Request for url: https://iot.seeed.cc/v1/user/login`

I checked everyhing, but still in troubles with the request

`r = requests.post('https://iot.seeed.cc/v1/user/login', data={'xxxxx@gmail.com':'xxxxxxx'}, verify=True)`

Any guess?
Thanks,
Peter

awong1900, 2016-07-26:
Hi Peter,
Thank you feedback.
I will fix it.

airdissy, 2016-07-27:
Hi, 
Thanks a lot, now it works and I could retrieve my token manually also.
It now say, server's running online.
However, (it's first time I use server_lean), I try to connect from mobile app.
When setup my Ip adress and login/password, it just say "Sorry".
I however see the ask for connection in the log.
Thanks,
Peter

airdissy, 2016-07-27:
Do I need to block 8080 port? by installing ufw
I'm not on esp8266 but on a mac so?

KillingJacky, 2016-07-27:
Hi airdissy, 
The lean server is only used for local data exchanging, it can not be used as the login server. You can use your deployed lean server with the following steps,
- Login the App to where you registered your Wio devices
- Select one of your Wio devices, click the 3-dots at the up-right corner, select "Device Setting"
- Switch on "Use Custom Server", and configure "Customized server" like this example: http://192.168.x.x:8080

That's it. Now you can monitor the logs on your server lean, it should receive heart beat from Wio device now.

airdissy, 2016-07-27:
Hi KillingJacky,

Thanks for the update. I was following the wiki about grove driver creation for Wio, maybe I get confused.
So when compiling my new driver on my server_lean folder, I should see it in the grove proposition on the mobile app after using custom server as you describe :
Process : 
/scan_drivers.py
/build_firmware.py
and finally
/server_lean.py

After connecting to mobile app with exchange on my wio, I should see it then?
Thanks !
Peter

KillingJacky, 2016-07-27:
Hi Peter,
Now I know that you want to create a grove driver :) To create a grove driver, you need not to touch server_lean stuff. The process will be,
- clone the git repo https://github.com/Seeed-Studio/Wio_Link (don't forget to update the submodule "grove_drivers" as well)
- write the driver, following this guide https://github.com/Seeed-Studio/Wio_Link/wiki/How-to-write-module-driver-for-Wio-Link%3F
- after making sure the driver is working, fork this repo https://github.com/Seeed-Studio/Grove_Drivers_for_Wio, commit your driver to your own forked repo
- pull a merge request
- we will check your driver and merge if it looks correct.
- wio users will see your contributed driver after our next time server update.

airdissy, 2016-07-27:
HI, 
Ok thanks for all of this. I will need some days to check this out and would come back.

airdissy, 2016-08-15:
Ok I could setup server and test my first drivers.
However, I have some difficulties, I will make a specific post.
Thanks,

closed 2016-08-15 by airdissy