Seeed-Studio/Wio_Link #30: Error with ./scan_drivers & xtensa-lx106-elf-gcc

airdissy, 2016-07-27:
Hello, 
I run on mac os X.
When running ./sac_drivers, i have the following error at the end  :
`/bin/sh: xtensa-lx106-elf-gcc: command not found
make: *** [aes.o] Error 127`

Then I installed xtensa-toolchain from github but nothing change?
Thanks,
Peter

awong1900, 2016-07-27:
@airdissy refer to http://stackoverflow.com/questions/14671021/bash-make-command-not-found

airdissy, 2016-08-01:
Hello, 
It's working now but I needed to restart the setup from beginning.
For those with same troubles with Mac setup. Here is the two tutorial I followed 
1) Install all command tool 
https://www.topbug.net/blog/2013/04/14/install-and-use-gnu-command-line-tools-in-mac-os-x/
2) Setup Esp-Open-sdk for mac as below, with especially a "command sensitive" filesystem
https://github.com/pfalcon/esp-open-sdk
Peter

closed 2016-08-01 by airdissy