Seeed-Studio/Wio_Link #31: scan_drivers python script fails when start directory contains .git directory

9037568, 2016-07-30:
When I run scan_drivers.py after cloning the repository, this error occurs:
```
=========
[]

ERR: can not find class file of X:\projects\electronics\wiolink\github\Wio_Link\grove_drivers\.git
```

The program should ignore directories such as this.

airdissy, 2016-07-30:
Hi,
Did you also clone the grove drivers directory, 
`git submodule init`
`git submodule update`

I'm not export, but I refer to below pages:
http://www.seeed.cc/project_detail.html?id=1273&utm_source=allwio&utm_medium=mailchimp&utm_campaign=Molly0628

9037568, 2016-07-30:
Yes, I cloned the drivers directory. But I did it separately using "git clone".

KillingJacky, 2016-08-03:
@9037568 Yes, that's a bug. Thank you!

KillingJacky, 2016-08-03:
Hotfixed, will close this :)

closed 2016-08-03 by KillingJacky