Seeed-Studio/Wio_Link #32: How to scan I2C module to get adress?

airdissy, 2016-08-03:
Hello,
I'm working on my first driver : grove dust sensor for which I need to know the I2C address.
I can't find a similar hex adress on a similar driver since no one work on a I2C driver for this sensor.
How to scan I2C device on my wio link to know the I2C adress of my grove module?
I found I2C scanner but only working on arduino. Is there are any way to launch a python program on my mac OS? I have already a copy of the server connected to my wio.
Other question : Is this hex adress similar for anyone with same grove module.
What happend when using the I2C hub?

Thanks for help.
Peter

KillingJacky, 2016-08-04:
Hi Peter,
Are you using this grove dust sensor? http://www.seeedstudio.com/Grove-Dust-Sensor-p-1050.html If so, this sensor is not a i2c device. In addition, this sensor needs 5V to work rather than 3.3V from the Grove socket of Wio Link.

airdissy, 2016-08-15:
Hello,
Maybe I would try later to power with external 5V.
Regarding correct port, what should be the correct port to drive it? (Digital, Uart?)
Thanks,

KillingJacky, 2016-08-25:
Contact with you in email, so close this.

closed 2016-08-25 by KillingJacky