Seeed-Studio/Wio_Link #33: add swagger.io api description to server.py

dothebart, 2016-08-12:
You've got a nice api description, however it always will list your server. 
If one tries to cut'n'paste the examples one has always to replace the hostname.

A swagger.io api description could be shipped alongside the server.py and be served by it. You then can talk to your local server.py, and directly test stuff.

Aside of that you can generate client stubs directly from the API.
We use swagger.io with ArangoDB for our API, and get lots of positive feedback for it: https://raw.githubusercontent.com/arangodb/arangodb/devel/js/apps/system/_admin/aardvark/APP/api-docs.json

You can also use the swagger editor with the compacter yaml representation for interactive editing.

I've tried to hook my wio link to my own server.py, but currently struggling with the procedure of authenticating and getting data from it?

KillingJacky, 2016-08-12:
If you deploy a private server, the hostname can be configured in config.py, with this done, the hostname in api description can be displayed as your configuration.

dothebart, 2016-08-12:
Its still somewhat problematic to get the curl strings from the device to the main machine (which is Why I think its a good idea to also have the documentation there) or is there a better way to get these commandlines other than typing it into the terminal one by one?

KillingJacky, 2016-08-12:
you can use the share icon up-right corner to send the ref page to computer.

closed 2016-08-25 by KillingJacky