Seeed-Studio/Wio_Link #34: pitfalls setting up own server

dothebart, 2016-08-12:
I'll list a bunch of pitfalls I trial and errored through, so maybe they may be fixed or documented for other users. 

I'm using a temparature sensor, to have a simple test target. 
I first set up a wio link with it, signed up to the public wio link server, configured my local wlan with it, got the device configured on my android. Let it compile the firmware for the sensor, uploaded it, were able to fetch values, everything fine.

Now the torture starts - wanted to have my own local server running. Fell into the trap that I followed:

https://github.com/Seeed-Studio/Wio_Link/wiki/Server-Deployment-Guide#33-how-to-tell-the-node-to-connect-the-lean-server

since it seemed to be the only howto telling how to connect to a local server. Made it re-compile and upload the firmware, saw that it wouldn't recognize the device. Used Curl and the api documentation to generate a User, and the device. Patched in to provision the device node, so it would recognize the board. But then, the server wouldn't accept data from the device:

 [I 160812 12:06:51 server:432] xchange device server accepted conns: 1
[I 160812 12:06:51 server:136] accepted sn: 3b4c4bf11918fad3671d5936f3108398 @fw_version 1.2
[E 160812 12:06:51 server:186] 'signature not match: \xc2\xd3<\xd6\xe6\x85P;(R\x15|\x95H#\x89\xd0\x03vR\x1a^\x95\\~8\xa8\xf3H\xa3\x9b\xc2 \xc4Q\rl}\xcb\x9e\xe9|\xc9T\xfe+E\xe1;\x14\xaf\x0f\xd7Q\x8c\xfe\xda\xee\x05|UmJ\x1fq'
[I 160812 12:06:51 server:437] xchange device server will remove connection: <__main__.DeviceConnection object at 0x7fae5e129710>

Since the app didn't allow to change the server, I unwinded everything, and then signed up to my local server. However, it wouldn't allow me to sign in with my previously created user & password, I had to create a new one.

Then I tried to re-add the wio-link. It failed with the state where the mobile tries to send the wio the wlan password, (which it seems it already knew) since I think these log-lines are from the communication with the wio-link:
[I 160812 12:40:06 web:1971] 200 GET /v1/nodes/list (192.168.3.121) 1.41ms
[I 160812 12:40:06 server:432] xchange device server accepted conns: 1
[I 160812 12:40:06 server:136] accepted sn: 85517c7c04a30c058501175447906bcb @fw_version 1.1
[I 160812 12:40:06 server:165] valid hello packet from node e5e5e380607811e685555c514fa4222f
[I 160812 12:40:06 server:429] ota device server accepted conns: 1
[I 160812 12:40:06 server:136] accepted sn: 85517c7c04a30c058501175447906bcb @fw_version 1.1
[I 160812 12:40:06 server:165] valid hello packet from node e5e5e380607811e685555c514fa4222f

However, the mobile would insist over and over again that it couldn't talk to the wio's hotspot when trying to send the wlan password. So the app wouldn't get into the state where it knows that there is a valid device configured with the server. 

So I pulled out my Wio Nano which hadn't been configured over here. I also had to re-push the connect button during the setup, but it eventually managed to get through that procedure. 

First time firmware compiling failed due to missing path environment, second time it failed somewhere inbetween, third time it worked.

So, it seems one can only change the server URL when working with a lean server? Please change the app so that it shows the OTA server URL in unchangeable, and below the changeable URL of the Data Exchange Server (lean server)

Please add a logic to the app that it scans with the Server whether the device already registered, if the device isn't there anymore when trying to set the wlan password.

KillingJacky, 2016-08-12:
Hi @dothebart , from your post above, I extract the following issues you have faced, if I missed something, please add on.
- don't know how to configure the App to communicate with a private server. Actually the App can do this. This is the step by step. 
    1) Log out your App by this operation procedure, setting - Log out
    2) At the login UI of the App, click "Server Location", then fill in your server address at the last option "Customize Server", e.g. http://your-ip-address:8080, then click "Save"
    3) Sign up and login

- can't re-add a Wio device. The failure has nothings to do with that the Wio device has been registered once. After entering the configuration mode by pressing the functional key, the Wio device does not give out messages, it just waits to be written the new configuration, it's totally passive. The failure is known to be a bug of the Android App. We have been trying to fix this, here is a test version which is still under the verification of Google Play, you can try downloading it from https://www.pgyer.com/wiolink.

We're sorry for the mismatch of the documentation. Will update the docs ASAP.  Thank you for your detailed usage report.

dothebart, 2016-08-12:
Please also change the 'device setting' dialog as I sugested above if I do get it right that this setting is only intended for lean servers, right?

Will there be a way to have several server accounts?

KillingJacky, 2016-08-12:
yes, honestly that's a little bit confusing. in the previous versions of the App, the text in that dialog gives a better expression that it's configuring the data exchange server. i didn't know this detail of this version as the UX was redesigned by someone else. i will get in touch with the UX designer to discuss these text expressions.

KillingJacky, 2016-08-12:
I couldn't get your meaning by "several server accounts". Does that mean creating several account in your local server?

dothebart, 2016-08-12:
either that, or several servers with different devices associated to it... The app itself gets all its state from the server, right?

dothebart, 2016-08-12:
If not, I'd at least like to be able to re-connect to my other account (maybe)

KillingJacky, 2016-08-12:
when logging into SEEED's servers, the authentication is sent to SEEED's bazaar user portal. If passed the authentication, an access token will be returned, which can be used to access API endpoints at Wio servers. Same account on different servers will have different device list.

dothebart, 2016-08-12:
yes, but the app can only handle one login at once, and if I want to change it, it will forget the other - right? And it seems as if the app always tries to create a new account? So if the server is on my laptop, and I carry that over into another network, which I may even have other WIO nodes, I'm in a very uncomfortable situation of having to start it all over again, right?

closed 2016-08-25 by KillingJacky