Seeed-Studio/Wio_Link #35: architecture question of xchange server path

dothebart, 2016-08-12:
Reading through the documentation I didn't find any notes on the communication flows. 
Reading through the code and watching netstat / tcpdump it seems to be like this:

1) the provisioned device connects the xchange server, and explains it which endpoints it has
2) this tcp connection device <-> xchange server is held open
3) an http client connects to one of the endpoints of the REST server
3a) the server queries the device via the existing tcp connection
3b) hopefully the device will reply with a value 
3c) the received value is transfered back to the REST HTTP-Client in json format

did I get that correctly?

KillingJacky, 2016-08-12:
yes, the same is another keep alive tcp connection, which connects device and OTA server. Device will listen OTA command, if triggered the device will download the firmware from an URL of OTA server.

dothebart, 2016-08-12:
is there a way to have the device push its values?

KillingJacky, 2016-08-12:
technically there is. but we followed the concept of web of things from W3C, in which a thing has properties to be read/written, and also events to push up. you can read some introduction from W3C.

dothebart, 2016-08-12:
Ah, so you're speaking about: 
https://www.w3.org/WoT/IG/wiki/Main_Page
and this one in special:
http://w3c.github.io/wot/architecture/wot-architecture.html
maybe you could add this somewhere in the wiki?

closed 2016-08-12 by dothebart