Seeed-Studio/Wio_Link #37: Debug serial not working?

dothebart, 2016-08-21:
I've tried to follow this howto: 
https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced-User-Guide#4-debug-console

with my Wio Link. I can see the startup messages via the USB interface, and as stated in the howto it stops talking after the reboot.
I've attached a serial usb converter to the UART1 as stated later on in the howto.

I saw the same boot prompt on that as on UART0 via the onboard USB serial interface - so it seems to be working.

Now I've changed `node_main/user_config.h` like that:
+#define ENABLE_DEBUG_ON_UART1       1

The only ready made example which I found in the source seems to be the sound example that has:

#if ENABLE_DEBUG_ON_UART1
	Serial1.print("interval");

http://www.seeedstudio.com/depot/images/product/Grove%20Speaker_01.jpg

So I've hooked up the speaker to the middle port next to the serial connector, re-flashed and was hoping to see 'interval' printed on the serial console - nothing happened.

 Whats wrong?

KillingJacky, 2016-08-25:
@dothebart You did so good catching up our new features. The debug serial enabling method has been changed recently, and I haven't managed to schedule my time to sync the docs yet. Now I have to finish this job, I will update the docs tonight. In short, you can enable the debugging serial via the [wio-cli](https://github.com/Seeed-Studio/wio-cli) tool after put the Wio device into setup mode.

KillingJacky, 2016-08-25:
Hi @dothebart , please check the new howto https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#4-debug-console

dothebart, 2016-08-25:
Is there a smart way to configure the wio shell from the state of the mobile client? It seems I failed to remember the passvoid..

dothebart, 2016-08-25:
I'm still fiddling to find a smart usecase combining the wios with ArangoDB, i.e. graphs https://github.com/arangodb/example-datasets#debian-dependency-graph since I like both a lot ;-)

KillingJacky, 2016-08-25:
Hi @dothebart you can try to retrieve your password here http://www.seeedstudio.com/find_psw.html . and here's a little intro to the user systems http://seeed-studio.github.io/Wio_Link/#user-management-apis

KillingJacky, 2016-08-25:
I noticed your active contributions to the ArangoDB, it's armed with pretty new concept! I want to have a closer investigation after I clean my messy backlogs a little bit 😂

KillingJacky, 2017-07-19:
Will close this, feel free to reopen it if any further question.

closed 2017-07-19 by KillingJacky