Seeed-Studio/Wio_Link #38: Fatal Exception (0)

blynch, 2016-09-01:
After discharging on a battery, the board will not wake up again.  I've tried flashing it twice. It will not accept config signal from the on-board switch or is it affected by a reset.  This happens with the battery or direct connect via usb to my laptop. The following statements are streaming to the serial console without end: 

Fatal exception (0): 
epc1=0x4020d310, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x0
0000000
Fatal exception (0): 
epc1=0x4020d310, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x0
0000000
Fatal exception (0): 
epc1=0x4020d310, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x0
0000000

blynch, 2016-09-01:
Additional message printed just after pressing reset: 

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
 
load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0
 
2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

KillingJacky, 2016-09-01:
Hi @blynch , seems like corruption is produced during your flashing the firmware. You can try the backdoor I left to turn over the firmware to the fail-safe partition. From the streaming of your serial console, I assume that your Wio unit will reboot itself after a while (caused by the watch dog). If this is right, then you just press and hold down the `CONFIG` button and wait the Wio unit reboot itself. When it reboots abnormally with `CONFIG` button pressed down at the same time, the firmware will try to roll back to the fail-safe partition.

blynch, 2016-09-02:
Hey @KillingJacky, thanks for the response!  It doesn't appear to get past the initial boot or restart on its own.  Holding down config has no affect along with pressing it repeatedly.  Tried flashing it again to no avail. It does not appear to roll back to the fail-safe partition. Is there a debug version of the firmware that could yield more information?

KillingJacky, 2016-09-02:
Hi @blynch If it cant pass the initial boot, the debug message from the firmware cant be printed then, as the serial cant be initialized. The messages you see at the serial console is printed by the bootloader. You said you tried flashing it again, what kind of flashing have you done? with ESP flash tool? If so, which files have you flashed? user1.bin and user2.bin ? or user1.bin, user2.bin and bootloader and a few more files like described in this [guide](https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link) ?  I think if the hardware is not damaged, flashing all the firmware related binary files will fix it, like a factory process.

blynch, 2016-09-08:
@KillingJacky I followed the steps in the [guide](https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link) on my mac using the  ESP python tool. It included user1.bin, user2.bin, esp_init_data_default.bin, blank.bin and the bootloader.  I don't think the hardware is damaged.  Do you have an updated bootloader I can test with?

KillingJacky, 2016-09-08:
The bootloader in this guide https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link is the working one. It's not the bootloader caused the case. If you flashed these 5 binaries and still no avail, I will doubt that the SPI flash may be damaged somehow. I have ever encountered such a failed unit which is fixed by the replacement of the SPI  flash.

juansezoh, 2017-01-13:
Hello, I have the same problem with a WioNode. I already use the erase_flash command and reload the bin files and it still does not work. Is there a command to return to the factory settings ?. The erease_flash erases all partitions ?. I appreciate what you can help me, thank you.

awong1900, 2017-01-16:
@juansezoh Now, no reset command to return to the factory settings. If your wio device is broke, you must update the firmware. That initial firmware of WioLink and WioNoe is the same. Please follow [this](https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#5-repair-bricked-wio-link) on Windows, or [this](https://github.com/Seeed-Studio/Wio_Link/wiki/Advanced%20User%20Guide#6224-flash-the-firmware) on Mac/Linux

closed 2017-04-27 by awong1900