Seeed-Studio/Wio_Link #40: Bad Format of parameter data_server

tube0013, 2016-10-05:
I have the server deployed locally, and my wiolinks show up as online, but when I tap the View API, I get:

Bad Format of parameter data_server.  SHould be: http[s]://domain-or-ip-address[:port]

I can query my the server and get data out with curl

awong1900, 2016-10-08:
Can you give me your server url than can use share to get.

tube0013, 2016-10-08:
http://my.domain/v1/node/resources?access_token=XXXXXXXXXXXXXXXXXXXXXXXXXX&data_server=https%3A%2F%2Fmy.domain

I edited out the token, and modified my domain name to not share my site publicly

tube0013, 2016-10-08:
looks like the %3A%2F%2F is not decoding to :// a if I change it manually I can access the api page.

awong1900, 2016-10-08:
It look like app's error. Do you use android or IOS?

tube0013, 2016-10-08:
IOS

awong1900, 2016-10-09:
@tube0013 we will check it.

KillingJacky, 2016-10-10:
Hi @tube0013 I have dug into the tornado source code, the tornado framework will handle the url decoding for query strings in the background. That's saying, after this call https://github.com/Seeed-Studio/Wio_Link/blob/dev/handlers.py#L1066, we get `self.vhost_url_base` as a decoded one like `https://your-domain`. What's your Python version and your tornado version? Make sure your python version >= 2.6 and tornado version == stable (4.4.2). If the versions are ok, please try to add one line per this position https://github.com/Seeed-Studio/Wio_Link/blob/dev/handlers.py#L1065 to your project file, and see in your log what does the `print` print?
```python
print self.request.arguments
```

tube0013, 2016-10-11:
okay I only had tornado 4.2 something, but got that updated.  still getting the error in the app.  I added the print statement to the handlers.py file.  Where does the log file write to by default/where can I configure it?  I tryed running server.py from the terminal, and see output, but don't see any thing related to the url when even when I try the View API from the app.

KillingJacky, 2016-10-12:
Run server.py from the terminal like this
`server.py --logging=debug`
when you browse your API page, you will get the console scrolling like the following 
```shell
[D 161010 10:27:41 handlers:597] node token:cb4eeac776----f48d130661830a5091
[D 161010 10:27:41 handlers:613] get current node, id: dbf0b----fd211e6bda61c1b0d14d127, name: Wio Link 3
{'access_token': ['cb4eeac776cb----130661830a5091'], 'data_server': ['https://192.168.31.239']}
[I 161010 10:27:41 handlers:1119] echo the cached page for node_id: dbf0b6617------da61c1b0d14d127
[I 161010 10:27:41 web:1946] 304 GET /v1/node/resources?access_token=cb4eeac776cb9xxxxxxx30a5091&data_server=https%3A%2F%2F192.168.31.239 (192.168.31.239) 2.73ms
```
Take note of this piece of json string `{'access_token': ['cb4eeac776xxxxxxxx661830a5091'], 'data_server': ['https://192.168.31.239']}`, you should see the `data_server` has already been decoded.

tube0013, 2016-10-12:
thank you, here is the result.  the `data_server` has not been decoded:

`{'access_token': ['6ec2ebfebed4547e662c0z32t8086c35'], 'data_server': ['https%3A%2F%2Fmy.domain']}
`
here are the versions of the python libraries:

>Name: tornado
>Version: 4.4.2

>Name: tornado-cors
>Version: 0.6.0

>Name: PyJWT
>Version: 1.4.0

>Name: PyYAML
>Version: 3.11

awong1900, 2016-10-13:
This is strange, I test it, url must be decode.
Now, Can you add more print below the line.
```
print self.request.uri
```
I suspect that the input url has been encode twice.

tube0013, 2016-10-13:
After adding the print line, the result:

`/v1/node/resources?access_token=6ec2ebfebed4547e662c0z32t8086c35&data_server=https%253A%252F%252Fmy.domain`

KillingJacky, 2016-10-13:
from this post 
[image] I assume that the URL sent out from your phone is correct. There must be something in front of the tornado server that encodes `%` again (into `%25`). Do you use any kind of web server as a frontend proxy?

tube0013, 2016-10-13:
yes using apache like in 1.1 of the Server Deployment guide.

```
BrowserMatch "MSIE [17-9]" ssl-unclean-shutdown

ProxyRequests Off
ProxyPreserveHost On
<Proxy *>
        Order deny,allow
        Allow from all
</Proxy>

ProxyPass /v1/node/event ws://127.0.0.1:9089/v1/node/event
ProxyPassReverse /v1/node/event ws://127.0.0.1:9089/v1/node/event

ProxyPass /v1 http://127.0.0.1:9089/v1
ProxyPassReverse /v1 http://127.0.0.1:9089/v1

```

awong1900, 2016-10-13:
Oh, this should be ok.  However, We can't repeat this bug on my side.
Can you test it on our Global Server with your phone? I hope can find some different.
And please tell me your phone model and ios version.

tube0013, 2016-10-13:
Would prefer to not have to reconfigure my wiolinks as they are working and reporting data to my Home-Assistant instance.   I have an iPhone 6 with iOS 10.0.2

KillingJacky, 2016-10-13:
I dont know where but I doubt the problem is with your apache configuration or apache version.  see here http://stackoverflow.com/questions/4390436/need-to-allow-encoded-slashes-on-apache 
Try adding `nocanon` to the end of every `ProxyPass` directive of the Wio related configuration.

KillingJacky, 2016-10-13:
Apache must have changed the default escape strategy for mod_proxy and mod_rewrite after a particular version, my deployment uses apache 2.4.10, the default works well.

tube0013, 2016-10-13:
I tried the `nocanon` with no luck.  I'm running 2.4.18.

this isn't huge as I'm able to get the url path and edit it to get to the api pages, but obviously not the desired experience.

KillingJacky, 2016-10-14:
So strange your apache🤔 No worry we still have the killer solution...change the python script
```python
        #print self.request.arguments
        #print self.request.uri
        self.vhost_url_base = self.get_argument("data_server", self.vhost_url_base)
        self.vhost_url_base = self.vhost_url_base.rstrip('/')
        self.vhost_url_base = escape.url_unescape(self.vhost_url_base)  ##add this line here
        #print self.vhost_url_base
```

tube0013, 2016-10-14:
well that works!  thank you.

closed 2016-10-27 by awong1900