Seeed-Studio/Wio_Link #41: Initial setup - 1042 if mobile data is turned on

nicolaevladescu, 2016-12-24:
Hello.

I was getting started with Wio Link and i found an issue when mobile data is also turned.
[Screenshot on Dropbox](https://www.dropbox.com/s/bua1wdbzuo3qwof/2016-12-23%2014.55.06.png?dl=0)

I was using a Nexus 5X with stock Android version 7.1.1 and  December 5, 2016 patch level.
After turning off mobile data i could successfully configure my Wio Link.

Also i noticed that Location must be turned on, i can't understand why it would need my location.
The error displayed if Location is turned off is also missleading.
[Screenshot on Dropbox](https://www.dropbox.com/s/wfijplxrvlxx9go/2016-12-23%2014.53.57.png?dl=0)

awong1900, 2016-12-24:
Hi @nicolaevladescu  Thank you for your support.
>  was using a Nexus 5X with stock Android version 7.1.1 and December 5, 2016 patch level.
After turning off mobile data i could successfully configure my Wio Link.

I will check it ASAP.

> Also i noticed that Location must be turned on, i can't understand why it would need my location.
The error displayed if Location is turned off is also missleading.

Because scan Wi-Fi network must open location service above Android 6.0 .
From android:
"App on Marshmallow needs ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission and Location services should be enabled to get wifi scan results"

nicolaevladescu, 2016-12-24:
Hello.

Thanks for the explanation regarding the Location question.

As for the Mobile Data issue, it's not exactly a bug because Android does not detect internet connectivity when you are connect to the Wio Link AP, and i would not invest hours in code or workarounds if it is not simple.

I would rather document it in the Getting Started guide because it is a one time think and not such a big deal for the user.

Thanks!

awong1900, 2016-12-26:
Thank you for your advise.
Yes, this look like the new feature of Android, which automatically detects network connectivity.

closed 2017-04-27 by awong1900