Seeed-Studio/Wio_Link #44: Moto G 4 Plus WiFi permission

sabas1080, 2017-02-08:
Hi 

I have Moto G 4 Plus and the app always asks me to check the wifi permissions, although the permissions granted appear

If you can pass that application window, try to connect with the link I get error 1042

Android version 6.01

Thanks

sabas1080, 2017-02-08:
I have done the GPS mentioned in #41, but I still have the error 1042

sabas1080, 2017-02-08:
I am testing with another device and get error 1044

It looks like the wio link can not connect to the server

awong1900, 2017-02-09:
@sabas1080 Can you tell me the blue led status when the phone display 1044? If the led is blink once per second, means the device is online. If it blink twice quickly, means can't get ip address, you can check your router's password.

sabas1080, 2017-02-11:
with error 1042, the led blnk very low, in mode setup

awong1900, 2017-02-13:
Than means the device did not receive the information from APP.  Make sure your phone had connect to Wio's device hot spots, f the automatic connection fails, please connect manually. Then click the Wio's SSID on the step.

KillingJacky, 2017-07-19:
Will close this, feel free to reopen it if any further question.

closed 2017-07-19 by KillingJacky