Seeed-Studio/Wio_Link #45: Error while deploying server in host filesystem

Shai1436, 2017-02-18:
Hi, I am trying to deploy the server into host file system. While executing "python ./scan_drivers.py" it generates error : xtensa-lx106-elf-gcc is not found. But previous commands executed successfully.
The step "xtensa-lx106-elf-gcc -v" generates "gcc version 4.8.2 ".

Also what is the download path of http://arduino.esp8266.com/package_esp8266com_index.json?
Thanks.

awong1900, 2017-02-19:
In the tools label, you can find the download link with your system. e.g. Ubuntu 12.04 use 
```
"tools": [
        {
          "version": "1.20.0-26-gb404fb9-2", 
          "name": "xtensa-lx106-elf-gcc", 
          "systems": [
            {
              "url": "http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz", 
              "checksum": "SHA-256:10476b9c11a7a90f40883413ddfb409f505b20692e316c4e597c4c175b4be09c", 
              "host": "i686-mingw32", 
              "archiveFileName": "win32-xtensa-lx106-elf-gb404fb9-2.tar.gz", 
              "size": "153527527"
            }, 
            {
              "url": "http://arduino.esp8266.com/osx-xtensa-lx106-elf-gb404fb9-2.tar.gz", 
              "checksum": "SHA-256:0cf150193997bd1355e0f49d3d49711730035257bc1aee1eaaad619e56b9e4e6", 
              "host": "x86_64-apple-darwin", 
              "archiveFileName": "osx-xtensa-lx106-elf-gb404fb9-2.tar.gz", 
              "size": "35385382"
            }, 
            {
              "url": "http://arduino.esp8266.com/osx-xtensa-lx106-elf-gb404fb9-2.tar.gz", 
              "checksum": "SHA-256:0cf150193997bd1355e0f49d3d49711730035257bc1aee1eaaad619e56b9e4e6", 
              "host": "i386-apple-darwin", 
              "archiveFileName": "osx-xtensa-lx106-elf-gb404fb9-2.tar.gz", 
              "size": "35385382"
            }, 
            {
              "url": "http://arduino.esp8266.com/linux64-xtensa-lx106-elf-gb404fb9.tar.gz", 
              "checksum": "SHA-256:46f057fbd8b320889a26167daf325038912096d09940b2a95489db92431473b7", 
              "host": "x86_64-pc-linux-gnu", 
              "archiveFileName": "linux64-xtensa-lx106-elf-gb404fb9.tar.gz", 
              "size": "30262903"
            }, 
            {
              "url": "http://arduino.esp8266.com/linux32-xtensa-lx106-elf.tar.gz", 
              "checksum": "SHA-256:b24817819f0078fb05895a640e806e0aca9aa96b47b80d2390ac8e2d9ddc955a", 
              "host": "i686-pc-linux-gnu", 
              "archiveFileName": "linux32-xtensa-lx106-elf.tar.gz", 
              "size": "32734156"
            }, 
            {
              "url": "http://arduino.esp8266.com/linuxarm-xtensa-lx106-elf-g46f160f-2.tar.gz", 
              "checksum": "SHA-256:f693946288f2ffa17288ef75ae16fa08573993f2b0a2a5e6bc35a68dc6087443", 
              "host": "arm-linux-gnueabihf", 
              "archiveFileName": "linuxarm-xtensa-lx106-elf-g46f160f-2.tar.gz", 
              "size": "34938475"
            }
          ]
        }, 
```

Shai1436, 2017-02-22:
Hi, In the first step you have asked to download the pre-compiled toolchain from [http://arduino.esp8266.com/package_esp8266com_index.json](here). But where to store this file in local filesystem.

awong1900, 2017-02-23:
You can see below:
e.g. Ubuntu 12.04 (so as the following steps)

cd /opt
wget http://arduino.esp8266.com/linux64-xtensa-lx106-elf-gb404fb9.tar.gz
tar -zxvf linux64-xtensa-lx106-elf-gb404fb9.tar.gz
sudo vim /etc/profile
#append the following line
export PATH=$PATH:/opt/xtensa-lx106-elf/bin
Then save the file, logout and login, test if the PATH is set correctly by inputing:

xtensa-lx106-elf-gcc -v

Shai1436, 2017-03-03:
Thanks for the reply awong1900. 
When I am starting the process, by running the command "sudo supervisorctl start esp8266" from the /opt/Wio_Link directory, It gives an error: no such process found. All the previous commands got executed without any errors.
Should I cd to another directory to run the command?

awong1900, 2017-03-03:
I search error "no such process found" on google. Can you post exact error message or image?

Shai1436, 2017-03-03:
Yes, the message in the console is    "esp8266: ERROR (no such process)"

awong1900, 2017-03-03:
Can you try "sudo service supervisord reload"? [http://stackoverflow.com/questions/19216042/error-no-such-process-nginxgunicornsupervisord](url)
Maybe we should add this to wiki document.

Shai1436, 2017-03-03:
Ok, I ran "sudo service supervisord reload", It gave output "supervisord: unrecognized service"

awong1900, 2017-03-03:
Can you use "supervisorctl tail <APP_NAME>" to get more access log?

Shai1436, 2017-03-03:
After running "supervisorctl tail esp8266", It gave output 
supervisor: couldn't chdir to /root/esp8266_iot_node/: ENOENT
supervisor: child process was not spawned

Also there is no folder named esp8266_iot_node in the root directory in my filesystem.

awong1900, 2017-03-03:
The file "esp8266_server.conf" is example. You can modify it with your system, make sure correct direction.

Shai1436, 2017-03-04:
Did you mean this variable "environment=PATH="/opt/xtensa-lx106-elf/bin:%(ENV_PATH)s" ". 
What values should I use to replace this with?

awong1900, 2017-03-08:
@Shai1436 if you manually run server.py is ok, you should check the supervisor config to make sure all item is correct. like dir path.

Shai1436, 2017-03-10:
Thanks @awong1900 , It worked eventually.

closed 2017-03-10 by Shai1436

awong1900, 2017-03-13:
you are welcome.