Seeed-Studio/Wio_Link #46: 'SERVER ERROR' Cannot Login/Register Android App

befreestudios, 2017-05-04:
Trying to set up everything, but continue to get vague 'SERVER ERROR' notification. Something wrong with the android app? Is there a web-app to use instead? Below are the symptoms and phone specs, Thanks in advance...

1) Downloaded android app.
2) I already have a Seeed account so tried to login, 'SERVER ERROR'
3) Tried to create an account, just in case, and still... 'SERVER ERROR'
4) Continue to try... nothing

-Pixel XL
-Android 7.1.5

awong1900, 2017-05-05:
Hi, first, last day i try resolved a domain certificate issue, the server have a few hours down. Can you try again today.

Use https://wio.seeed.io/login, you can test your account validity. 
And we have CLI tools [wio](https://github.com/Seeed-Studio/wio-cli)
This tool can help to setup device.

dtev, 2017-05-05:
Having same issue with iOS app and web login using both credentials or Facebook auth

awong1900, 2017-05-05:
I confirm this issue.I need classmate to check seeed auto service. Please wait a few hours.

befreestudios, 2017-05-05:
@awong1900 Thanks for the quick response. I tried what you mentioned above and when I hit that url and enter my credentials I get a https://http.cat/404. If I have time I'll try CLI I will (I was planning on it anyway), but curious whats going on here... let me know if you need anything else from me

thx

befreestudios, 2017-05-05:
@awong1900 I just tried this again and was able to login through the app now. This seems to have been resolved for me.

dtev, 2017-05-05:
Still getting connection error

___________________________
Michele (Mike) Hjorleifsson
IT and Media Architect 
PH: 772-224-8913

awong1900, 2017-05-08:
@dtev @befreestudios  the auto server is fixed. Can you try again?

dtev, 2017-05-08:
Still getting connection errors on signup or login

___________________________
Michele (Mike) Hjorleifsson
IT and Media Architect 
PH: 772-224-8913

awong1900, 2017-05-09:
@dtev Please uninstall wio app if you use ios. And install again.

dtev, 2017-05-09:
I was able to login after removing and reinstalling the iOS app. Will test further and advise

___________________________
Michele (Mike) Hjorleifsson
IT and Media Architect 
PH: 772-224-8913

awong1900, 2017-05-09:
Thanks for the feedback. sorry for the inconvenience.

Thunderhorsellc, 2017-05-28:
Same IOS login issues. After trying everything mentioned above, I registered under another email address and it worked.

awong1900, 2017-05-28:
Please test login on this page https://wio.seeed.io/login.

Thunderhorsellc, 2017-05-28:
OK, but worked with other after I used a different email address. Used ssme
here.

closed 2017-05-31 by awong1900