Seeed-Studio/Wio_Link #47: Wio Link Doesn't connect to private server

kp372, 2017-05-25:
I have setup a private server on aws as explained here.
[](https://github.com/Seeed-Studio/Wio_Link/wiki/Server-Deployment-Guide)

Installation worked smoothly from above guide. But somehow my wio link device doesn't connect to the server. I know this from the blue led blink once quickly then off 1s suggesting its in connecting to the server state.

I have tried to connect my wio link to seeed server and it works well. Do you have any idea which I can try?

awong1900, 2017-05-26:
Can you confirm the AWS enable  the 8000 port?  As wio device connect server with TCP port.
below code is about port info.
```
    app = myApplication(conn, cur)
    http_server = HTTPServer(app, xheaders=True)
    http_server.listen(8080)

    app2 = myApplication_OTA(conn, cur)
    http_server2 = HTTPServer(app2)
    http_server2.listen(8081)

    tcp_server = DeviceServer(conn, cur, 'xchange')
    tcp_server.listen(8000)

    tcp_server2 = DeviceServer(conn, cur, 'ota')
    tcp_server2.listen(8001)
```

kp372, 2017-05-26:
Thank You for your reply. Great, it is working now!

closed 2017-05-26 by kp372