Seeed-Studio/Wio_Link #50: Multichannel Gas Sensor ist not working

dynamik1703, 2017-07-14:
Hello,

I'm trying to use the Multichannel Gas Sensor with the Wio.

But it's not possible for me. The test request returns: {"error": "The sensor is not connected."}

I have two Wios and already the second sensor. No combination was working.

Is there a bug?

Best wishes

KillingJacky, 2017-07-16:
When did you purchase your multichannel gas sensor? In the mean time I will check with colleague  if the sensor has been revised.

dynamik1703, 2017-07-16:
I bought it 2-3 weeks ago. It's the v1.0.

Every reguest gives the response: {"error": "The sensor is not connected."}

The only exception is the CO reguest. The response is: {"concentration_ppm":-1.0}

KillingJacky, 2017-07-17:
The sensor's firmware has been updated to version 2 which breaks the protocol. You might bought the new version. I will fix the Wio driver. Sorry for the inconvenience.

dynamik1703, 2017-07-17:
Thanks great! Can you please tell me when it is ready and what do I have to do to update it?

Best wishes

KillingJacky, 2017-07-18:
I will ping you here when it's done. What you need to do is simple, just update the firmware for your Wio via the App.

mlilius, 2017-07-18:
I have this issue as well. Thank you for addressing this.

KillingJacky, 2017-07-19:
Fixed with 75440ce35ecfbca9c79029efa1804287c7fa6662
Some notes about this sensor:
- The version 2 firmware (the gas sensor) will not respond any i2c request in the first 1 second at startup, because the LED blink pattern needs 1 second. So please read the sensor after 1 second.
- The gas sensor needs to be heated at least 1 minute to get the right reading. So please read the sensor after 1 minute the Wio power up.

mlilius, 2017-07-19:
Thank you. I can confirm that this works now. Thanks again.

dynamik1703, 2017-07-20:
Works for me as well! Awesome.

Thank you very much. Great service!

KillingJacky, 2017-07-21:
Pleasure. Thank you guys for reporting as well. 
Will close this, if any further questions please reopen.

closed 2017-07-21 by KillingJacky

GGaannLLaayy, 2019-03-06:
I couldn't use Multichannel gas sensor with ESP32..There is no result in serial print ...I am using Grove expansion board especially for ESP32..Thanks...