Seeed-Studio/Wio_Link #51: Deploy a Local Lean Data Exchange Server into Raspberry Pi

NalinSilva, 2017-07-31:
Hello everyone,
I am new to this environment and sorry for any inconvenience. I am trying to deploy a lean server into Rpi and still could not do that. When start the lean_server it show the message on cmd as follows.
Please help me to solve this issue
[image]

KillingJacky, 2017-07-31:
You should use python 2.

NalinSilva, 2017-07-31:
Thanks for quick response. I'll try with python 2 Thanks

NalinSilva, 2017-07-31:
is it ok with python2.7.
when i run 'pip install pycrypto'  terminate the instalation process and show
[image]

NalinSilva, 2017-07-31:
hello Sir,
 pycrypto installed successfully after installing python-dev. Now "concurrent : 344" error msg is missing with python 2.I'am trying  to connect Wio app with lean server now.I'll give update asap when done it.
Thanks for you supporting.

NalinSilva, 2017-08-04:
Hi,Sorry for the late, I had to move on another task immediately.Now i got following msg on Rpi
[image]

but still i could not connect node via lean server using wio app.when i try i got response as undefined in wio app

NalinSilva, 2017-08-04:
Done now i can connect lean server using wio app.when i give the data excahange server IP with '' http://''  i got the data of sensor which is connected to the wio node.Anyware thanks for help to Mr. KillingJacky.

closed 2017-08-04 by NalinSilva