Seeed-Studio/Wio_Link #54: 404 on login from andriod app

rovincent, 2017-08-18:
when trying to login from android i get 404 in the lean_server output and connection error on the phone.

luminosa-servers, 2017-08-18:
we suddenly starting having trouble with the WIO app on all 3 of our android phones. Using our own server which is responding to the CLI, so I am not sure what is happening. Tried reinstalling android app but with the same result. It connects then immediately shuts down with "sorry...there is a problem.' error message.

luminosa-servers, 2017-08-18:
Interesting fact is that when the android phone is not connected to any WiFi, the WIO app does not fail. Of course it cannot communicate with any device. The moment I connect to a wifi signal with internet, the wio app dies.

luminosa-servers, 2017-08-28:
I discovered that a bad board (may a bad sensor on the board) was causing the program to terminate. I eliminated the board (which was the last one I installed) and everything is now working. 

I will eventually learn what caused the problem was but the sensor is an analog buzzer. The reason I thought it may be the problem was because I could not get the API to respond.

KillingJacky, 2017-09-11:
@rovincent Could you give us a screenshot about the 404 error if you still have issues with logging in?

@luminosa-servers Glad to know that you already have it working.

closed 2024-10-15 by baorepo