Seeed-Studio/Wio_Link #55: How to onboard wio link without wio mobile app?

richacker, 2017-09-06:
I am trying to create a web solution for wio link where I can sign up and login. Further, while creating a node, using following code:

curl -H"Authorization: token aKleqQebIsjmQMSCokemtgNXMD4Y7OTe3uyzEL1iwwA"-d name=node_1 -d board="Wio Link v1.0"https://us.wio.seeed.io/v1/nodes/create

{

   "node_key":"e3d3b0fe920ddc3c69393fdbf83a2114",

   "node_sn":"819554bd9519c94a19cc7fbd32447118"

}

I am unable to link the newly created board to the physical board kept with me. When I onboard new board on the app, it is added as another board which works fine but the board created through API remains offline forever. How do I create a wio link board and get it working without mobile app?

KillingJacky, 2017-09-07:
When creating a device, the last step is writing the retrieved node_sn and node_key into the Wio device. This is done via UDP. Now the App is doing this thing. To build a web solution, the browser has no capability to send UDP packages, a standalone proxy application or browser plugin is needed to transform the network communication from websocket to UDP.
Another way to write the private key is using a usb cable, write through the Serial interface of the Wio device.
Would you like to drop an email to me (xuguang.shao@seeed.cc) if you still have interests? I can explain more details to you and maybe help some in writing the private key to device.

KillingJacky, 2017-09-15:
Will close this issue and turn into email loops.

closed 2017-09-15 by KillingJacky