Seeed-Studio/Wio_Link #58: Recommended procedure to setup and use Wio Link without Seeed app and servers

armijnhemel, 2018-01-17:
hello,

someone already asked part of the question in #55 but I was wondering: what is the recommended procedure to setup and use the Wio Link without using the Seeed app and servers?

I know I can run my own server and change the server in the app after registering with Wio's servers, but  is there information on how to do everything locally without connecting to Wio's servers? Documentation would be very much appreciated.

thanks!

KillingJacky, 2018-01-22:
Hi @armijnhemel , Wio Link can be working without the App, but it cant work at all if it doesn't connect to a server - no matter the server is a Seeed's public sandbox server or a private deployed server.

armijnhemel, 2018-01-22:
I understand. What I meant is that I also want to be able to configure the Wio Link devices without first registering with the Seeed servers (I mistyped there and said "Wio's servers", apologies). How can I do that?

KillingJacky, 2018-01-23:
😆 I got you. Please refer to here https://github.com/Seeed-Studio/Wio_Link/wiki/Server-Deployment-Guide#41-how-to-use-the-full-private-server and this youtube video https://youtu.be/qIdtKa0-xvs

armijnhemel, 2018-01-27:
I knew about that part ^_^

But my question is about the following: how can I configure the Wio Link devices without needing to sign into any Wio server?

If I run the Wio app I need to login on a Wio server. If I use wio-cli I first need to login on a Wio server as well. I guess that by attaching a serial cable I should be able to set the right information on a Wio Link as well, without the need to login to a Wio server, but how?

KillingJacky, 2018-01-29:
To send raw command flow is kinda hacking behavior, there will not be any official documentation about that. But the source code is open, and you can refer to the wio-cli tool, e.g. https://github.com/Seeed-Studio/Wio-Cli/blob/ce83f4c2d30be7f72d1a128acd123dfc5effa563/wio/commands/cmd_udp.py, it's a good point that can help to inspect the low level command flow

armijnhemel, 2018-01-29:
Thanks, this should get me further :-)

closed 2024-10-15 by baorepo