Seeed-Studio/Wio_Link #59: How to use Wio Node and upload custom code via OTA/Serial?

akiyamaerik, 2018-03-24:
Hello Guys! 

I am new to Wio and i'm looking for some help. I've bought Wio Node hoping that it would be a plug-and-upload code microcontroller, since I wanted to code in C and upload to this little beast.

However, I am facing here some difficulties. Wio Node does not get detected by windows when I plug it into a USB port, and I do not know if it is possible to upload a code OTA

Could you guys please help me? I'm looking on the Seeed page but I-ve found nothing so far.

Thanks in advance!!

KillingJacky, 2018-03-26:
Hi @akiyamaerik Wio Node has no USB-to-Serial convertor, it can't be used as plug-n-upload board. The micro-USB port of Wio Node is used only for power.

akiyamaerik, 2018-03-26:
@KillingJacky but is it even possible to upload a custom code to Wio? For example, via UART?

KillingJacky, 2018-03-26:
Yes it's possible. You need to setup 2 boot pins before you power up the board. Please refer to ESP8266's datasheet.

geiseri, 2020-11-20:
@KillingJacky is there a way to pack an OTA update to reflash it?  I am using the wio-link with ESPHome and it's awesome.  I have a few nodes that would fit nicely into that.

kylehase, 2021-04-22:
@geiseri I found a way to flash ESPHome or Tasmota to a Wio Node, in case you're still looking.

1. Connect a USB-TTL to the grove D0 connector Tx/Rx/GND pins. Don't connect the grove VCC.
2. Connect microUSB (for power only)
3. While holding the FUNC button, press and release the RST button to enter flash mode
4. Flash ESPHome or Tasmota as you normally would through USB-TTL (for example using ESPHome Docker over USB or Tasmotizer)

If the ESP8266EX is not detected, try swapping Tx/Rx pins.

closed 2024-10-15 by baorepo