Seeed-Studio/Wio_Link #68: Issues connecting Wio Node to new self hosted server

dcflachs, 2020-04-23:
I am having some real trouble getting my Wio Nodes connected to a self hosted server (docker). I am using the wio-cli as i cant manage to get the android apps to find the Wio Configuration Access Point. Via wio-cli i can connect to the node but i always fail at 
```
>> The Wio connect to Server failure.

> Please check info you enter, Try again.
```
looking at the server logs i can see that the wio node connection was apparently successful.
```
[I 200423 13:29:56 server:149] accepted sn: 25c49bc204dcb73023f1d65de907e095 @fw_version 1.1
[D 200423 13:29:56 server:170] sig:     b77545a058f68e62707fcccd65c639b4ec6cef9d2e4a830a4348a27c5614a3f5
[D 200423 13:29:56 server:171] sig calc:b77545a058f68e62707fcccd65c639b4ec6cef9d2e4a830a4348a27c5614a3f5
[I 200423 13:29:56 server:179] valid hello packet from node c90e00d4858711eaa8e80242ac140006
[I 200423 13:29:56 server:188] >>>>>>>>>>>>>>>>>>>>>
[I 200423 13:29:56 server:189] channel: xchange
[I 200423 13:29:56 server:190] size of conn pool: 1
[I 200423 13:29:56 server:191] <<<<<<<<<<<<<<<<<<<<<
[D 200423 13:29:57 server:204] cipher text: 26f8bc2d84e60f90064416b6851d1f18959df3a3cf1d3928fbd43fb43eef629c
[I 200423 13:29:57 server:149] accepted sn: 25c49bc204dcb73023f1d65de907e095 @fw_version 1.1
[D 200423 13:29:57 server:170] sig:     b77545a058f68e62707fcccd65c639b4ec6cef9d2e4a830a4348a27c5614a3f5
[D 200423 13:29:57 server:171] sig calc:b77545a058f68e62707fcccd65c639b4ec6cef9d2e4a830a4348a27c5614a3f5
[I 200423 13:29:57 server:179] valid hello packet from node c90e00d4858711eaa8e80242ac140006
[I 200423 13:29:57 server:188] >>>>>>>>>>>>>>>>>>>>>
[I 200423 13:29:57 server:189] channel: ota
[I 200423 13:29:57 server:190] size of conn pool: 1
[I 200423 13:29:57 server:191] <<<<<<<<<<<<<<<<<<<<<
```

I also see 

```
[D 200423 13:33:57 server:336] heartbeat sent to node c90e00d4858711eaa8e80242ac140006 on xchange channel
[D 200423 13:33:57 server:336] heartbeat sent to node c90e00d4858711eaa8e80242ac140006 on ota channel
[D 200423 13:33:57 server:252] Node c90e00d4858711eaa8e80242ac140006 alive on xchange channel!
[D 200423 13:33:57 server:256] Node c90e00d4858711eaa8e80242ac140006 recv json on ota channel
[D 200423 13:33:57 server:257] {u'msg': 1, u'msg_type': u'online_status'}
```

At this point i am not sure where to direct my debugging.

awong1900, 2020-04-26:
Thank you for using. I am the developer of cli. I will check you issue.

awong1900, 2020-04-26:
Do you have login for wio-cli? Because you use yourself server.
```
(env2) jump@wangtengdeMacBook-Pro ~ % wio login
1.) Global Server (New)[https://us.wio.seeed.io]
2.) Chinese Server [https://cn.wio.seeed.io]
3.) Customize Server
? Please choice server: 3
? Please enter Customize Server url
(e.g. https://us.wio.seeed.io or http://192.168.1.10:8080): 

```

dcflachs, 2020-04-26:
@awong1900 Thanks for taking a look. The server i am using for my devices is not currently internet accessible. Rather i had intended to use all the Wio devices and the server on the same private network. Are you asking for access to the server for debugging purposes?

awong1900, 2020-04-27:
Whether you use a private network or a public network, you need to make sure that wio-cli has access permissions  to the servers you set up. The `wio-cli login` command gets the token from your server to configure the operation.

dcflachs, 2020-04-27:
Ok I understand what you are asking now. Yes i was able to log in to the private server using the wio-cli login command. The wio state command returns the correct server, email and a token. My attempts to connect the node to the server using the wio setup command always fail as mentioned above. 

Just a note i did also try using the android app. I was able to connect that to the private server as well but i could never connect to the wio node config access point with android. I think that was because of my android tablet using 5G wifi.

dcflachs, 2020-04-30:
@awong1900 I am sorry i did not mean to confuse the issue by mentioning 5G wifi. I do have a dual band router and i am connecting my Wio devices to the 2.4G wifi band as they do not support 5G wifi. When i am using the wio-cli i am using it with a 2.4G PC wireless adapter. 

My comment above that mentioned 5G wifi was in regards to an issue i had with the Wio android app. It is not directly relevant to this thread since i am using wio-cli to connect the nodes via PC.

awong1900, 2020-04-30:
OK.   This is my connection result just now, compare it, if can help you.
```
(env2) jump@wangtengdeMacBook-Pro env2 % wio setup
> Setup is easy! Let's get started...

! PROTIP: Hold the Configure button ~4s into Configure Mode!
! PROTIP: Please make sure you are connected to the Server

? Would you like continue? [Y/n]: 
0.) Wio Link v1.0
1.) Wio Node v1.0
? Please choice the board type: 1
                                                                                 
! PROTIP: Wireless setup of Wio!
! PROTIP: You need manually change your Wi-Fi network to Wio's network.
! PROTIP: You will lose your connection to the internet periodically.

? Please connect to the Wio_* network now. Press enter when ready: 
? Would you like to manually enter your Wi-Fi network configuration? [y/N]: 
0.) nuVision_2.4G                                                                
1.) CHOD
2.) nuVision_2.4G
3.) ChinaNet-MaAG
4.) SZCH
5.) 朗诗集团
6.) CHGLBhuiyishi
7.) CMCC
8.) and-Business
9.) TP-LINK_4BBB
10.) ChinaNet-q75W
11.) ChinaNet-AqDH
12.) jasson-bangongshi
13.) CMCC
14.) RopaiSamba
15.) 360行车记录仪-8A83
? Please select the network to which your Wio should connect: 0
> Please enter your Wi-Fi network password (leave blank for none): 55557777
> Please enter the name of a device will be created: ten-node
> Here's what we're going to send to the Wio:

> Wi-Fi network: nuVision_2.4G
> Password: 55557777
> Device name: ten-node

? Would you like to continue with the information shown above? [Y/n]: 

> The Wio connect to Server success.                                             
> Set Wio name success.                                                          
                                                                                 
> Configuration complete!
(env2) jump@wangtengdeMacBook-Pro env2 % 
```

awong1900, 2020-04-30:
I'm guessing a few possible problems, such as domain name errors, or HTTPS without certificates.

dcflachs, 2020-04-30:
The workflow you posted above looks generally the same as what i am seeing right up until your says `The Wio connect to Server success. `
Mine prints 
```
>> The Wio connect to Server failure.
> Please check info you enter, Try again.
```

The domain name that i am using is https://192.168.0.10:447
which is the address of a reverse proxy to a Wio_Link server running in a docker container. The other 3 port addresses used by the Wio_Link  server are not reverse proxies (and not ssl protected). They are set as described in the server deployment guide 8081, 8000, and 8001.

The certificate that I am using is self signed.

awong1900, 2020-04-30:
I'm looking at the code on for wio-cli. [here](https://github.com/Seeed-Studio/wio-cli).
First send with upd command:
```
    elif board == WIO_NODE_V1_0:
        r = upd_send(msvr, msvr_ip, xsvr, xsvr_ip, node_sn, node_key)
        if not r:
            return
        d_name = r['name']
        check_connect(mserver_url, token, node_sn, d_name)
```
Then check wio list 
```
def check_connect(mserver_url, token, node_sn, d_name):
    thread = termui.waiting_echo('')
    thread.daemon = True
    thread.start()
    state_online = False
    for i in range(60):
        thread.message("The Wio now attempt to connect to Server... [%s]" %(60-i))
        time.sleep(1)
        try:
            params = {"access_token":token}
            r = requests.get("%s%s" %(mserver_url, node_list_endpoint), params=params)
            r.raise_for_status()
            json_response = r.json()
        except Exception as e:
            continue

        for n in json_response["nodes"]:
            if n["node_sn"] == node_sn and n["online"]:
                click.echo(click.style('\r> ', fg='green') + "The Wio connect to Server success.              ")
                thread.message("Setting Wio name...")
                state_online = True
                break
        if state_online:
            break

    if not state_online:
        thread.stop('')
        thread.join()
        click.echo(click.style('\r>> ', fg='red') + "The Wio connect to Server failure.")
        click.secho("\n> Please check info you enter, Try again.", fg='white', bold=True)
        return None
```
The problem lies somewhere in between. It could be an upd delivery failure or some parameter setting error.

dcflachs, 2020-04-30:
Do you also have access to logs from your WIO_Link server? In my first post in this thread I have logs from my private server that seem to show Wio_Link server communicating with my Wio Node. That was what made me post here initially as i think it suggests an issue with the way i have my server setup rather than the Wio-CLI or the node itself.

awong1900, 2020-04-30:
we need @KillingJacky help. :)

awong1900, 2020-04-30:
wio-cli can enable the node's debug mode. Should have more log on server.
```
Config your device setting:

$wio config --debug [on|off], enable/disable wio debug
$wio config --get-debug, get wio debug status

awong1900, 2020-04-30:
> Do you also have access to logs from your WIO_Link server? 
I don't have the permissions anymore.

dcflachs, 2020-05-19:
I am still very much looking for some help with this one. @KillingJacky

dcflachs, 2020-05-31:
I was finally able to sit down and get to the bottom of this issue. @awong1900 it turns out that you were correct and the issue was with wio-cli and no the Wio_link server. I have logged an [issue](https://github.com/Seeed-Studio/wio-cli/issues/13) in the wio-cli repository. 

**The TLDR for anyone landing at this issue is that Wio-cli was not playing nice with a the self signed certificate on my custom server.**

closed 2020-05-31 by dcflachs