Seeed-Studio/Wio_Link #75: Server Error Message

HamadMoniem, 2021-04-24:
Im just a normal user who knows nothing about iot but I do know that the app just sucks and all I want is to make the led strip turn on.  Every time I try to sign in with the correct credentials, try to sign in with google, make a new account with 10 differant emails, it says 'server error'. My server location is Global.. How do I fix this issue, I just want to sign in please, thank you

MichaelUMB, 2021-04-28:
@HamadMoniem  - tracking this - I haven't used my Wio Link in a while, and can't open the app because I get "connection error" (with both iphone Xs and ipad). I tried a private server using https://myIPaddress and that did not work either.

ryazwinski, 2023-03-22:
yea same.. seeed hosted servers are dead.. built a private server, installed a real cert, still no dice.. 
unfortunately seeed seems to have abandoned this product line and support is non-existent ...  I've tried the iphone app, and cli neither works.
if anyone has solutions, I'd love to hear.

ryazwinski, 2024-10-15:
abandoned would be more accurate than "completed"
big FU to the seeed supporters who funded these projects